platformio/platform-ststm32 #215: Does not find files (see attached log)

renejeanmercier, 2019-03-05:
Processing development (framework: arduino; platform: https://github.com/platformio/platform-ststm32.git; board: bluepill_f103c8_128k)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8_128k.html
PLATFORM: ST STM32 > BluePill F103C8 (128k)
HARDWARE: STM32F103C8T6 72MHz 20KB RAM (128KB Flash)
DEBUG: CURRENT(stlink) EXTERNAL(blackmagic, jlink, stlink)
Warning! Cannot find linker script for the current target!

Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <MMicro_Rotary_Encoder>
|-- <Ucglib> 1.5.2
|   |-- <SPI> 1.0
Compiling .pioenvs/development/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pioenvs/development/FrameworkArduino/HardwareTimer.cpp.o
Compiling .pioenvs/development/FrameworkArduino/IPAddress.cpp.o
In file included from /home/rjm/.platformio/packages/framework-arduinoststm32-maple/STM32F1/cores/maple/HardwareSerial.h:39:0,
from /home/rjm/.platformio/packages/framework-arduinoststm32-maple/STM32F1/cores/maple/HardwareSerial.cpp:33:
/home/rjm/.platformio/packages/framework-arduinoststm32-maple/STM32F1/cores/maple/boards.h:39:10: fatal error: board.h: No such file or directory

***************************************************************
* Looking for board.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:board.h"
* Web  > https://platformio.org/lib/search?query=header:board.h
*
***************************************************************

#include <board.h>                                                                            //#include <board/board.h>
^~~~~~~~~
compilation terminated.
Compiling .pioenvs/development/FrameworkArduino/Stream.cpp.o
/home/rjm/.platformio/packages/framework-arduinoststm32-maple/STM32F1/cores/maple/HardwareTimer.cpp:32:10: fatal error: board/board.h: No such file or directory
#include <board/board.h>           // for CYCLES_PER_MICROSECOND

====================================

[env:development]
platform = https://github.com/platformio/platform-ststm32.git
board = bluepill_f103c8_128k
board_build.core = maple
framework = arduino
upload_protocol = dfu
upload_port = /dev/ttyACM0

===================================

Thank you for your help,

Regards,

RJM

pfeerick, 2019-03-09:
A suitable linker script is missing for this target... 

https://github.com/platformio/platform-ststm32/blob/8fdf59a24b4a77dd984ec98379ebe491e82f1d11/boards/bluepill_f103c8_128k.json#L17

Some poking around in the stm32duino configuration files indicates that the linker FLASH value for that particular configuration should be 131072 (versus the 65536 / 64K of the ... well, 64K board!). 

https://github.com/stm32duino/Arduino_Core_STM32/blob/c7dffcd33fb285cb66881ed97e8772f8398614c6/boards.txt#L588

renejeanmercier, 2019-03-18:
Hi Peter,

Please tell me where this information should be placed (directory,
filename, etc.) ...Could not find any clear documentation on this matter on
the PIO site.

Thank you for your help.

Regards,

R.J. Mercier
===================================

pfeerick, 2019-03-18:
I'm not sure just yet. Keep an eye on any updates on #224 and have a look at the [last few entries on this post](https://community.platformio.org/t/stm32f1-blue-pill-stuck-in-dfu-mode-after-upload/6853/24?u=pfeerick)... There are some hackish ways to do it, but as part of #224 I'm trying to work out how to get it fixed properly...

valeros, 2019-03-18:
Hi @renejeanmercier! If I'm not mistaken there is no 128K version of `bluepil` in the maple core? As a workaround, you can try to use `genericSTM32F103CB` target.

renejeanmercier, 2019-03-18:
Hi Valerii!

There is one but it is buggy. It is called <bluepill_f103c8_128k>.

You are an AS !!!! With your suggestion, the program compiles without error and warning and executes perfectly ... like it does with Arduino.

Thank you so much and PLEASE pass the word around.

**This line is for Peter**. Peter, the parameters of  the < genericSTM32F103CB > can be used for the bluepill 128k without modification.

Regards,

R.J.M.

closed 2019-03-18 by renejeanmercier

pfeerick, 2019-03-18:
Thanks for that @renejeanmercier ... it indeed does seem to work, so I'll just keep my issue open until I can figure out why it works when the linker file appears to be missing...

renejeanmercier, 2019-03-19:
Hi Peter,

Thank you for your follow-up. 

When a solution is found for the Bluepill 128k, if you don't mind, please send me a e-mail and share with Valerii who was  so kind to offer an alternative and a trail to a solution.

Regards,

Rene-Jean

====================