platformio/platform-ststm32 #217: STM32CubeMx Issues with the compilation under VSCode

gimi87, 2019-03-07:
Hello :)

I prepared a simple STM32F4DISCOVERY project sketch with STM32CubeMx software.
I only checked:
- Add necessary library files as reference in the toolchain project configuration file,
- Generate peripheral initialization as a pair of'.c/.h' files per peripheral.

Then i tried to compile under platformio in vscode. This is my platformio.ini file:

```
[platformio]
include_dir = Inc
src_dir = Src

[env:disco_f407vg]
platform = ststm32
board = disco_f407vg
framework = stm32cube
build_flags =
    -I$PIOHOME_DIR/packages/framework-stm32cube/f4//Middlewares/ST/STM32_USB_Host_Library/Class/CDC/Inc
    -I$PIOHOME_DIR/packages/framework-stm32cube/f4//Middlewares/ST/STM32_USB_Host_Library/Core/Inc
    -std=c++14
```

and this is result:

```
Processing disco_f407vg (platform: ststm32; board: disco_f407vg; framework: stm32cube)
-----------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/disco_f407vg.html
PLATFORM: ST STM32 > ST STM32F4DISCOVERY
HARDWARE: STM32F407VGT6 168MHz 192KB RAM (1MB Flash)
DEBUG: CURRENT(stlink) ON-BOARD(stlink) EXTERNAL(blackmagic, jlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs\disco_f407vg\src\gpio.o
Compiling .pioenvs\disco_f407vg\src\i2c.o
Compiling .pioenvs\disco_f407vg\src\i2s.o
Compiling .pioenvs\disco_f407vg\src\main.o
Compiling .pioenvs\disco_f407vg\src\spi.o
Compiling .pioenvs\disco_f407vg\src\stm32f4xx_hal_msp.o
Compiling .pioenvs\disco_f407vg\src\stm32f4xx_it.o
Compiling .pioenvs\disco_f407vg\src\syscalls.o
Compiling .pioenvs\disco_f407vg\src\system_stm32f4xx.o
Compiling .pioenvs\disco_f407vg\src\usb_host.o
Compiling .pioenvs\disco_f407vg\src\usbh_conf.o
Compiling .pioenvs\disco_f407vg\src\usbh_platform.o
Compiling .pioenvs\disco_f407vg\FrameworkBSP\stm32f4_discovery.o
Compiling .pioenvs\disco_f407vg\FrameworkBSP\stm32f4_discovery_accelerometer.o
Compiling .pioenvs\disco_f407vg\FrameworkBSP\stm32f4_discovery_audio.o
Compiling .pioenvs\disco_f407vg\FrameworkHALDriver\Src\stm32f4xx_hal.o
Src\usbh_conf.c: In function 'HAL_HCD_PortEnabled_Callback':
Src\usbh_conf.c:181:3: warning: implicit declaration of function 'USBH_LL_PortEnabled'; did you mean 'USBH_LL_GetToggle'? [-Wimplicit-function-declaration]
USBH_LL_PortEnabled(hhcd->pData);
^~~~~~~~~~~~~~~~~~~
USBH_LL_GetToggle
Compiling .pioenvs\disco_f407vg\FrameworkHALDriver\Src\stm32f4xx_hal_adc.o
Compiling .pioenvs\disco_f407vg\FrameworkHALDriver\Src\stm32f4xx_hal_adc_ex.o
Src\usbh_conf.c: In function 'HAL_HCD_PortDisabled_Callback':
Src\usbh_conf.c:191:3: warning: implicit declaration of function 'USBH_LL_PortDisabled'; did you mean
'USBH_LL_SetTimer'? [-Wimplicit-function-declaration]
USBH_LL_PortDisabled(hhcd->pData);
^~~~~~~~~~~~~~~~~~~~
USBH_LL_SetTimer
Compiling .pioenvs\disco_f407vg\FrameworkHALDriver\Src\stm32f4xx_hal_can.o
Compiling .pioenvs\disco_f407vg\FrameworkHALDriver\Src\stm32f4xx_hal_cec.o
[... 81 lines cut ...]
Compiling .pioenvs\disco_f407vg\FrameworkCMSISDevice\gcc\startup_stm32f407xx.o
Compiling .pioenvs\disco_f407vg\FrameworkCMSISDevice\system_stm32f4xx.o
Archiving .pioenvs\disco_f407vg\libFrameworkCMSISDevice.a
Archiving .pioenvs\disco_f407vg\libFrameworkHALDriver.a
Linking .pioenvs\disco_f407vg\firmware.elf
.pioenvs\disco_f407vg\src\usb_host.o: In function `MX_USB_HOST_Init':
usb_host.c:(.text.MX_USB_HOST_Init+0x8): undefined reference to `USBH_Init'
usb_host.c:(.text.MX_USB_HOST_Init+0x16): undefined reference to `USBH_RegisterClass'
usb_host.c:(.text.MX_USB_HOST_Init+0x22): undefined reference to `USBH_Start'
usb_host.c:(.text.MX_USB_HOST_Init+0x3c): undefined reference to `CDC_Class'
.pioenvs\disco_f407vg\src\usb_host.o: In function `MX_USB_HOST_Process':
usb_host.c:(.text.MX_USB_HOST_Process+0x2): undefined reference to `USBH_Process'
.pioenvs\disco_f407vg\src\usbh_conf.o: In function `HAL_HCD_SOF_Callback':
usbh_conf.c:(.text.HAL_HCD_SOF_Callback+0x4): undefined reference to `USBH_LL_IncTimer'
.pioenvs\disco_f407vg\src\usbh_conf.o: In function `HAL_HCD_Connect_Callback':
usbh_conf.c:(.text.HAL_HCD_Connect_Callback+0x4): undefined reference to `USBH_LL_Connect'
.pioenvs\disco_f407vg\src\usbh_conf.o: In function `HAL_HCD_Disconnect_Callback':
usbh_conf.c:(.text.HAL_HCD_Disconnect_Callback+0x4): undefined reference to `USBH_LL_Disconnect'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\disco_f407vg\firmware.elf] Error 1
===================================== [ERROR] Took 6.08 seconds =====================================
The terminal process terminated with exit code: 1
```

Can you help to compile the project ? :)

gimi87, 2019-04-16:
Attached the project

[Test.zip](https://github.com/platformio/platform-ststm32/files/3086469/Test.zip)

valeros, 2019-05-02:
Thanks, @gimi87! As for now, I'd recommend adding the USB source files as a standalone library in the `lib` folder. I modified your project, so you can try to build it by yourself.
[Test-updated.zip](https://github.com/platformio/platform-ststm32/files/3139847/Test-updated.zip)

mirao, 2019-05-24:
I had similar issue when I was compiling [default demo app](https://github.com/mirao/stm32f4discovery/tree/master/stm32cube-default-demo-app) for board STM32F407G-DISC1. The app source is distributed from STM32Cube_FW_F4 repository
I had to put necessary middleware files into `lib` folder and then building started working.

It would be great if STM32CubeMX allowed project's export to PlatformIO format out of the box (they already provide configuration for some other IDE's), but it's another story.

ivankravets, 2019-07-18:
> It would be great if STM32CubeMX allowed project's export to PlatformIO format out of the box (they already provide configuration for some other IDE's), but it's another story.

We work on that and hope ST Micro will add it soon.

Temporary solution is https://github.com/platformio/platform-ststm32/issues/217#issuecomment-488821662

closed 2019-07-18 by ivankravets