platformio/platform-ststm32 #218: framework-libopencm3 outdated

radek-senfeld, 2019-03-07:
Hello,
package framework-libopencm3 seems quite outdated. Any plans to update to current codebase? I could help but I'm not able to find the repo for this package.

kinosang, 2019-08-08:
It's versioned `16_10_2015`, it's too old for development.

Since libopencm3-example uses libopencm3/libopencm3@7be50a5e75, it's better to update the framework up-to-date.

kinosang, 2019-08-08:
[framework-libopencm3.zip](https://github.com/platformio/platform-ststm32/files/3482126/framework-libopencm3.zip)

I have create my own `2.0.0` package for `framework-libopencm3`, you can download and extract it into `~/.platformio/packages`.

* If you want to use it with official `platform-ststm32`, you should modify the `package.json` and change the `version` to `1.1.0`

DavidPowell, 2019-12-11:
As a clarification, the current version of libopencm3 packaged with platformio fails to compile many of the libopencm3 examples on github, such as this [USB CDC example](https://github.com/libopencm3/libopencm3-examples/blob/master/examples/stm32/f4/stm32f4-discovery/usb_cdcacm/cdcacm.c).

@kinosang Could you clarify what steps you took to generate your own version of the framework?

As far as I can tell, the platformio [builder script](https://github.com/platformio/builder-framework-libopencm3) for libopencm3 does not fetch a specific version of libopencm3 source, it just relies on having the source available locally. So there doesn't seem to be any way for users to submit a pull request that fixes this issue.

kinosang, 2019-12-15:
@DavidPowell just download the source code of libopencm3, put them in correct place of `framework-libopencm3`, and modify the Makefiles to avoid `script syntax` problem

tute-avalos, 2020-06-25:
> As a clarification, the current version of libopencm3 packaged with platformio fails to compile many of the libopencm3 examples on github, such as this [USB CDC [...]

That is what intrigues me. I can't figure out where PlatformIO gets the libopencm3 sources, it should be from the 'master' branch of the project on github, but no ... It's a mystery... 

I want to teach with this tools, so It would be nice to have the proper updates without making manual installations

tute-avalos, 2020-06-25:
I think I get it, PlatformIO gets the framework-libopencm3-1.1.0.tar.gz from https://sourceforge.net/projects/platformio-storage/files/packages/

valeros, 2020-08-03:
Resolved in https://github.com/platformio/platform-ststm32/commit/0bf8d98f1944d76e4b8159962c04969d610c30ab

closed 2020-08-03 by valeros