platformio/platform-ststm32 #220: Black Pill board variant - deliberate or copy/paste error?

pfeerick, 2019-03-09:
Just noticed when looking at the black pill boards files that the `variant` is `BLUEPILL_F103XX` for both of the F103 boards (but the other black_ boards seems fine)... is that a deliberate choice or a copy/paste error?  Happy to do a PR to correct that.

aster94, 2019-03-10:
Should be the same, the two F1 blue and black boards are exactly the same while the bigger board (F4 blue/black) are different. Really once existed also the F1 red pill but i think that there are not more on aliexpress 🤔

pfeerick, 2019-03-11:
Other than having a different pinout and that the builtin led is on different pins ;) Both the red and blue variants had it on PC13, but the black variant is on PB12 ... just to be different... :confused: But other than that minor detail, they are the same - programming-wise anyway.

aster94, 2019-03-11:
You are right i forgot about the built in led, maybe it is enough to requiere a different board variant!

valeros, 2019-05-09:
The `variant` field for this board is set according to the [original value in the framework](https://github.com/stm32duino/Arduino_Core_STM32/blob/master/boards.txt#L627).

closed 2019-05-09 by valeros