platformio/platform-ststm32 #221: Promblem with exapmle arduino-external-libs

Nullfati, 2019-03-10:
After uploading by STLink v2 on Blupill F103C8 and connect board via USB get an error "USB Device not recognized".
Operation system: Windows 10 x64

valeros, 2019-03-12:
Hi @Nullfati ! Thanks for reporting. Does it work with Arduino IDE?

Nullfati, 2019-03-12:
Hi @valeros.
Yes, on Arduino with [this framework](https://github.com/rogerclarkmelbourne/Arduino_STM32) (as I understand it's used maple wrappers, not HAL) works fine.
I investigate this problem and found out that PlatformIO default for Blue pill use STM32duino.
On STM32duino Serial doesn't work with an integral USB port.
I today received USB to TTL dongle (FTD1232) and will try to use it with a Blue pill and STM32duino framework. When getting some result about it, leave feedback in this article.

After some tests, I see the main flaw of usage maple wrapper (`board_build.core = maple`) than STM32duino, empty firmware with maple take >50% of program memory unlike STM32duino take only 12%~. For my case, it's very important to have space in program memory.

valeros, 2019-03-25:
So how do you see the solution to your problem? Is the firmware size smaller in Arduino IDE?

Nullfati, 2019-03-25:
Firmware size smaller in Arduino IDE.
I think need some additional libraries for this example for working USB port on BluePill F103C8.

valeros, 2019-03-27:
Could you please provide more info about the difference in sizes? How much smaller is the firmware produced by Arduino IDE?

fpistm, 2019-03-28:
Some notes about that

**_Target BluePill F103C8 bare sketch_**

# STM32 core (v1.5.0) without USB enabled
```
Sketch uses 8336 bytes (12%) of program storage space. Maximum is 65536 bytes.
Global variables use 852 bytes (4%) of dynamic memory, leaving 19628 bytes for local variables. Maximum is 20480 bytes.
```
# STM32 core (v1.5.0)  USB CDC enabled
```
Sketch uses 20568 bytes (31%) of program storage space. Maximum is 65536 bytes.
Global variables use 3776 bytes (18%) of dynamic memory, leaving 16704 bytes for local variables. Maximum is 20480 bytes.
```
# Maple core (top of the master)
```
Sketch uses 15128 bytes (23%) of program storage space. Maximum is 65536 bytes.
Global variables use 3136 bytes (15%) of dynamic memory, leaving 17344 bytes for local variables. Maximum is 20480 bytes.
```
I guess you do not enable the USB for the STM32 core that's why you get only 12%.
And so maple is better if you need less memory consumption, in general case maple has a better support for the BP as it is mainly designed for this while STM32 core is generic across 12 STM32 families.
Note that if you don't use USART it can be disabled. Then with only USB CDC:
```
Sketch uses 18788 bytes (28%) of program storage space. Maximum is 65536 bytes.
Global variables use 3416 bytes (16%) of dynamic memory, leaving 17064 bytes for local variables. Maximum is 20480 bytes.
```
In several advanced sketch, STM32 core size is less or equivalent than maple one as it use the newlib nano, maple don't.
Hope this is clear and help.

Nullfati, 2019-03-28:
Hi @fpistm,

Can you please provide your build configs?
I received some other results.

**STM32duino clear:**
platformio.ini:
```
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
```

main.cpp:
```
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

Result:
```
DATA:    [          ]   4.2% (used 852 bytes from 20480 bytes)
PROGRAM: [=         ]  12.6% (used 8252 bytes from 65536 bytes)
```

**Maple clear:**
platformio.ini:
```
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
board_build.core = maple
```

main.cpp:
```
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

Result:
```
DATA:    [==        ]  20.9% (used 4272 bytes from 20480 bytes)
PROGRAM: [===       ]  28.0% (used 18332 bytes from 65536 bytes)
```

**STM32duino with serial:**
platformio.ini:
```
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
```

main.cpp:
```
#include <Arduino.h>

void setup() {
  Serial.begin(115200);
  Serial.println("Start");
}

void loop() {
}
```

Result:
```
DATA:    [          ]   4.2% (used 852 bytes from 20480 bytes)
PROGRAM: [==        ]  17.1% (used 11184 bytes from 65536 bytes)
```

**Maple with serial:**
platformio.ini:
```
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
board_build.core = maple
```

main.cpp:
```
#include <Arduino.h>

void setup() {
  Serial.begin(115200);
  Serial.println("Start");
}

void loop() {
}
```

Result:
```
DATA:    [==        ]  20.9% (used 4272 bytes from 20480 bytes)
PROGRAM: [===       ]  28.1% (used 18436 bytes from 65536 bytes)
```

fpistm, 2019-03-28:
@Nullfati This is result from Arduino IDE not from pio.

closed 2021-02-16 by Nullfati