platformio/platform-ststm32 #222: HardwareTimer not available... what to use?

silardg, 2019-03-11:
I have been using HardwareTimer last year for a project.

Now it won't compile with the same code and I haven't found a solution.

I need the interrupt execution of a function every 10ms.

Any suggestions?

valeros, 2019-03-11:
Hi @silardg ! What board are you using? Probably you need to [switch to the maple core](http://docs.platformio.org/en/latest/platforms/ststm32.html#switching-between-arduino-cores).

silardg, 2019-03-12:
@valeros 

I have fixed it by switching to version 2.0.0 :/

I don't have time to debug it, but it works like this. 

I would suggest not closing this issue, maybe someone finds an easy workaround to having HardwareTimer.

closed 2019-03-23 by ivankravets