platformio/platform-ststm32 #223: Crash during build when framework is not defined

me21, 2019-03-13:
I'm building a bootloader for STM32 which doesn't use any framework. So my `platformio.ini` looks like

```
[env:esp32-autotag]
platform = ststm32
board = genericSTM32F103C8
build_flags = -DTARGET_GENERIC_F103_NO_LED_OR_BUTTON -Istm32_lib -Iusb_lib -Wl,-Tstm32_lib/c_only_md_high_density.ld -DXTAL12M
src_filter = +<*> -<sketch_combiner/> -<stm32_lib/c_only_startup_user.s>
```

Notice there's no framework set.

I get the following crash during build:

```
Error: Traceback (most recent call last):
  File "c:\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 107, in cli
    result = (envname, ep.process())
  File "c:\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 207, in process
    result = self._run()
  File "c:\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 313, in _run
    return p.run(build_vars, build_targets, self.silent, self.verbose)
  File "c:\.platformio\penv\lib\site-packages\platformio\managers\platform.py", line 353, in run
    self.configure_default_packages(variables, targets)
  File "C:\.platformio\platforms\ststm32\platform.py", line 28, in configure_default_packages
    if "arduino" in variables.get("pioframework") and build_core == "maple":
TypeError: argument of type 'NoneType' is not iterable
```

This happens in platform versions 5.2.0 and 5.3.0. Version 5.1.0 builds my firmware just fine.

valeros, 2019-03-18:
Thanks, should be fixed in the dev branch!

closed 2019-03-18 by valeros

aster94, 2019-09-20:
same problem here!

```
pio run
Processing d1 (platform: espressif8266; board: d1_mini_lite)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 
Error: Traceback (most recent call last):
  File "c:\users\vincenzo\.platformio\penv\lib\site-packages\platformio\__main__.py", line 90, in main
    cli(None, None, None)
  File "c:\users\vincenzo\.platformio\penv\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\vincenzo\.platformio\penv\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\users\vincenzo\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 37, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\vincenzo\.platformio\penv\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\vincenzo\.platformio\penv\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\vincenzo\.platformio\penv\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\users\vincenzo\.platformio\penv\lib\site-packages\click\decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\vincenzo\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 115, in cli
    silent, verbose, jobs, is_test_running))
  File "c:\users\vincenzo\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 135, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "c:\users\vincenzo\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 78, in process
    self.jobs)
  File "c:\users\vincenzo\.platformio\penv\lib\site-packages\platformio\managers\platform.py", line 380, in run
    self.configure_default_packages(options, targets)
  File "C:\Users\vincenzo\.platformio\platforms\espressif8266\platform.py", line 22, in configure_default_packages
    if "arduino" not in framework:
TypeError: argument of type 'NoneType' is not iterable

============================================================

An unexpected error occurred. Further steps:

  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

valeros, 2019-09-20:
@aster94 Fixed, thanks for reporting. Next time please open a new issue in the appropriate repository.

aster94, 2019-09-20:
Thanks @valeros 😀 I thought it was almost the same problem so i didn't opened a new one since i was just giving the suggestion of more verbose error messages for the users

commit 214c8811 (2019-03-18): Allow empty framework field // Resolve #223