platformio/platform-ststm32 #224: 128k flash variants of STMF103

pfeerick, 2019-03-15:
When looking at the F103, I realised the 128k board variant doesn't appear to work as the linker script is missing. Since I wanted to try to get it working, I checked how the stm32dunio core handled the 64/128k variants, and it seems that it is merely changing the FLASH line in the linker script to 128k. 

https://github.com/platformio/platform-ststm32/blob/a1c78e09e4a966a4240d9ad5ab1ff6c8103e47fe/ldscripts/stm32f103x8.ld#L4

.. creating a linker file as `stm32f103x8_128k.ld` and updating the path to point to it in ...
https://github.com/platformio/platform-ststm32/blob/a1c78e09e4a966a4240d9ad5ab1ff6c8103e47fe/boards/bluepill_f103c8_128k.json#L17

... should have been enough, but instead, I get a `Warning! Cannot find linker script for the current target!` and the build fails. What am I missing? Changing the `bluepill_f103c8.json` or `genericSTM32F103CB.json` and `stm32f103x8.ld` in place seems to work just fine though, and the generated code does appear to be functioning on a test board. It's just not the appropriate way to fix it! 

And on a semi-related note, I see that https://github.com/platformio/platform-ststm32/blob/develop/boards/genericSTM32F103CB.json points to the version of the linker file that specified 64k of flash, whereas both the chip and the `maximum_size` are for 128k, so that is completely wrong. 

If I can get the bluepill 128k working, I can do PRs for both it and the STM32F103CB.

fpistm, 2019-03-15:
FYI, 

To ease those kind of issue related to linker script. STM32 core now preprocess the linker file.
This allow to avoid linker duplication and customize it for FLASH/RAM size start address... using define.

https://github.com/stm32duino/Arduino_Core_STM32/blob/0038f9eb8dbdda09a0bbf73cde199118cc03fd66/variants/BLUEPILL_F103XX/ldscript.ld#L55
https://github.com/stm32duino/Arduino_Core_STM32/blob/0038f9eb8dbdda09a0bbf73cde199118cc03fd66/variants/BLUEPILL_F103XX/ldscript.ld#L63-L64

Related preproc commit:
Add preprocessing of linker script:
https://github.com/stm32duino/Arduino_Core_STM32/commit/8ab8f67e29b6231a5c40febe7ee59201661c933c

 Add flags to prelink recipe for linker script preprocessing:
https://github.com/stm32duino/Arduino_Core_STM32/commit/c43bb81da3b3b50a26898e7529720d8a8352eb64

neilbalch, 2019-03-16:
@fpistm I'm afraid your point isn't coming across. What are you trying to say?

pfeerick, 2019-03-16:
He is saying that for the stm32duino core, that the `-DLD_MAX_SIZE=` and `-DLD_MAX_DATA_SIZE=` build flags allow the ram and flash sizes to be specified those flags. I'm not sure though if that will work for platformio or when the maple core is used.

fpistm, 2019-03-16:
Maple does'nt preprocess linker file. They simply use 2 linker files. One with 64k, the other with 128k.
@neilbalch I only explain how this is handle in the STM core which is now provided with platformio.
This method ease linker file customization and avoid duplicate it. It's also fine if bootloader is used.
 I don't know how pio process build options so now, up to pio maintainer to choose how handle this issue  😉

valeros, 2019-03-18:
Hi @pfeerick! Thanks for reporting. I'm not sure that this board supports the maple core directly, but you can use `generic_stm32f103c` as the variant. All logic that operates on linker scripts you can find in the build script here: `.platformio/packages/framework-arduinoststm32-maple/tools/platformio-build-stm32f1.py`.

pfeerick, 2019-03-18:
Looks like I'll have to dig a bit deeper to see what's going on... as you pointed out in #215 , `genericSTM32F103CB` works out of the box as a 128k option. I did miss that it is pointing to a different linker file... `stm32f103x8.ld` vs `stm32f103xb.ld` ... just that last character, dang it! :laughing: ... but I don't know where it's pulling it from as it does compile fine. 

btw, this was when using `board_build.core = maple` to force usage of the maple core so DFU could be used...