platformio/platform-ststm32 #225: upload-reset is not a dynamic executable

nopnop2002, 2019-03-18:
$ uname -a
	Linux HP-Laptop 4.15.0-46-generic #49-Ubuntu SMP Wed Feb 6 09:33:07 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

	$ pio update
	Updating tool-scons                      @ 2.20501.7      [Up-to-date]

	Platform Manager
	================
	Platform ST STM32
	--------
	Updating ststm32                         @ a1c78e0        [Up-to-date]
	Updating tool-stm32duino                 @ 1.0.1          [Up-to-date]
	Updating framework-arduinoststm32-maple  @ 1.10000.190228 [Up-to-date]
	Updating tool-openocd                    @ 2.1000.190118  [Up-to-date]
	Updating framework-arduinoststm32        @ 3.10500.190225 [Up-to-date]
	Updating toolchain-gccarmnoneeabi        @ 1.70201.0      [Up-to-date]

	Library Manager

  $ ldd .platformio/packages/tool-stm32duino/upload-reset
	        not a dynamic executable

  $ .platformio/packages/tool-stm32duino/upload-reset
  -bash: .platformio/packages/tool-stm32duino/upload-reset: No such file or directory

pfeerick, 2019-07-13:
Reproducable on 
```
Platform ST STM32
--------
Updating ststm32                         @ 5.4.3          [Up-to-date]
Updating tool-stm32duino                 @ 1.0.1          [Up-to-date]
Updating framework-arduinoststm32-maple  @ 1.10000.190516 [Up-to-date]
Updating tool-openocd                    @ 2.1000.190422  [Up-to-date]
Updating framework-arduinoststm32        @ 3.10500.190327 [Up-to-date]
Updating toolchain-gccarmnoneeabi        @ 1.70201.0      [Up-to-date]
```

I think the underlying issue is that the wrong flavour binaries are being installed. On a 64bit Ubuntu 18.04 VM, 

```
uname -a
Linux ubuntu-hyperv 4.15.0-54-generic #58-Ubuntu SMP Mon Jun 24 10:55:24 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

file ~/.platformio/packages/tool-stm32duino/upload-reset 
/home/pfeerick/.platformio/packages/tool-stm32duino/upload-reset: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-, for GNU/Linux 2.6.24, BuildID[sha1]=dc8dc5ad6de94179a9d534b1a34aeb6d906640b1, not stripped

ldd ~/.platformio/packages/tool-stm32duino/upload-reset 
	not a dynamic executable
```

pfeerick, 2019-07-16:
I was able to resolve this issue by running `sudo apt-get install lib32z1` on the Ubuntu 18.04 LTS VM, but you may need a [different command if you are running a different version](https://support.humblebundle.com/hc/en-us/articles/202759400-Installing-32-bit-libs-on-a-64-bit-Linux-system).