platformio/platform-ststm32 #227: I can't build blink.cpp using platformio + STM32FreeRTOS

nopnop2002, 2019-03-22:
My environment.

	$ uname -a
	Linux HP-Laptop 4.15.0-46-generic #49-Ubuntu SMP Wed Feb 6 09:33:07 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

platformio.ini

	$ cat platformio.ini
	; PlatformIO Project Configuration File
	;
	;   Build options: build flags, source filter
	;   Upload options: custom upload port, speed and extra flags
	;   Library options: dependencies, extra library storages
	;   Advanced options: extra scripting
	;
	; Please visit documentation for the other options and examples
	; https://docs.platformio.org/page/projectconf.html

	[env:nucleo_f103rb]
	platform = ststm32
	board = nucleo_f103rb
	framework = arduino
	board_build.core = stm

src/main.cpp

	#include <Arduino.h>
	#include <STM32FreeRTOS.h>

	#define LED PB12

	void setup()
	{
	  // initialize LED digital pin as an output.
	  Serial.begin(115200);
	  Serial.println("Serial start");
	  pinMode(LED, OUTPUT);
	}

	void loop()
	{
	  digitalWrite(LED, HIGH);
	  delay(1000);
	  digitalWrite(LED, LOW);
	  delay(1000);
	}

pio run

	$ pio run
	Processing nucleo_f103rb (platform: ststm32; board: nucleo_f103rb; framework: arduino)
	--------------------------------------------------------------------------------
	Verbose mode can be enabled via `-v, --verbose` option
	CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_f103rb.html
	PLATFORM: ST STM32 > ST Nucleo F103RB
	HARDWARE: STM32F103RBT6 72MHz 20KB RAM (128KB Flash)
	DEBUG: CURRENT(stlink) ON-BOARD(stlink) EXTERNAL(blackmagic, jlink)
	Library Dependency Finder -> http://bit.ly/configure-pio-ldf
	LDF MODES: FINDER(chain) COMPATIBILITY(soft)
	Collected 8 compatible libraries
	Scanning dependencies...
	Dependency Graph
	|-- <STM32duino FreeRTOS> 9.0.3
	Compiling .pioenvs/nucleo_f103rb/src/main.cpp.o
	Linking .pioenvs/nucleo_f103rb/firmware.elf
	.pioenvs/nucleo_f103rb/lib6d3/libSTM32duino FreeRTOS_ID2093.a(heap.c.o): In function `__malloc_lock':
	heap.c:(.text.__malloc_lock+0x0): multiple definition of `__malloc_lock'
	/home/nop/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/libc_nano.a(lib_a-mlock.o):mlock.c:(.text.__malloc_lock+0x0): first defined here
	.pioenvs/nucleo_f103rb/lib6d3/libSTM32duino FreeRTOS_ID2093.a(heap.c.o): In function `__malloc_unlock':
	heap.c:(.text.__malloc_unlock+0x0): multiple definition of `__malloc_unlock'
	/home/nop/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/libc_nano.a(lib_a-mlock.o):mlock.c:(.text.__malloc_unlock+0x0): first defined here
	collect2: error: ld returned 1 exit status
	*** [.pioenvs/nucleo_f103rb/firmware.elf] Error 1
	========================== [ERROR] Took 4.22 seconds ==========================

ivankravets, 2019-03-25:
Could you try opening PlatformIO IDE Terminal and type
```
pio update
```

Try to build again. Does it work now?

nopnop2002, 2019-03-26:
OK
It was fixed.
Thank you.
Please close this issue.

closed 2019-03-27 by ivankravets