platformio/platform-ststm32 #229: STM32F401RE demo fails / linker file not found

robinvalk, 2019-03-29:
Hi,

I was currently following along with the following tutorial:
https://docs.platformio.org/en/latest/tutorials/ststm32/stm32cube_debugging_unit_testing.html#tutorial-stm32cube-debugging-unit-testing

While building the project the following error occurred:
```Scripts\platformio.exe run <

Processing nucleo_f401re (platform: ststm32; board: nucleo_f401re; framework: stm32cube)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_f401re.html
PLATFORM: ST STM32 > ST Nucleo F401RE
HARDWARE: STM32F401RET6 84MHz 96KB RAM (512KB Flash)
DEBUG: CURRENT(stlink) ON-BOARD(stlink) EXTERNAL(blackmagic, jlink)
Warning! Cannot find a linker script for the required board! Firmware will be linked with a default linker script!
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs\nucleo_f401re\src\main.o
Compiling .pioenvs\nucleo_f401re\FrameworkBSP\stm32f4xx_nucleo.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_adc.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_adc_ex.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_can.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_cec.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_cortex.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_crc.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_cryp.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_cryp_ex.o
Archiving .pioenvs\nucleo_f401re\libFrameworkBSP.a
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_dac.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_dac_ex.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_dcmi.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_dcmi_ex.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_dfsdm.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_dma.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_dma2d.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_dma_ex.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_dsi.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_eth.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_flash.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_flash_ex.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_flash_ramfunc.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_fmpi2c.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_fmpi2c_ex.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_gpio.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_hash.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_hash_ex.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_hcd.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_i2c.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_i2c_ex.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_i2s.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_i2s_ex.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_irda.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_iwdg.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_lptim.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_ltdc.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_ltdc_ex.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_mmc.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_nand.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_nor.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_pccard.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_pcd.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_pcd_ex.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_pwr.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_pwr_ex.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_qspi.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_rcc.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_rcc_ex.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_rng.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_rtc.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_rtc_ex.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_sai.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_sai_ex.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_sd.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_sdram.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_smartcard.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_spdifrx.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_spi.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_sram.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_tim.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_tim_ex.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_uart.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_usart.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_hal_wwdg.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_adc.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_crc.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_dac.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_dma.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_dma2d.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_exti.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_fmc.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_fsmc.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_gpio.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_i2c.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_lptim.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_pwr.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_rcc.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_rng.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_rtc.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_sdmmc.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_spi.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_tim.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_usart.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_usb.o
Compiling .pioenvs\nucleo_f401re\FrameworkHALDriver\Src\stm32f4xx_ll_utils.o
Compiling .pioenvs\nucleo_f401re\FrameworkCMSISDevice\gcc\startup_stm32f401xe.o
Compiling .pioenvs\nucleo_f401re\FrameworkCMSISDevice\system_stm32f4xx.o
Archiving .pioenvs\nucleo_f401re\libFrameworkCMSISDevice.a
Archiving .pioenvs\nucleo_f401re\libFrameworkHALDriver.a
Linking .pioenvs\nucleo_f401re\firmware.elf
arm-none-eabi-gcc: error: Valk\.platformio\packages\framework-stm32cube\platformio\ldscripts\STM32F401RE_DEFAULT.ld: No such file or directory
*** [.pioenvs\nucleo_f401re\firmware.elf] Error 1
===================================================================================== [ERROR] Took 7.19 seconds =====================================================================================
The terminal process terminated with exit code: 1````

I've also tried this example which gives the same result:
https://github.com/platformio/platform-ststm32/tree/develop/examples/stm32cube-hal-blink

Am I doing something wrong?

robinvalk, 2019-03-29:
I think I've found the problem. The ld script path was not surrounded in quotation marks and therefore a path containing a space was not interpreted correctly.

See: https://github.com/platformio/platformio-core/pull/2264

ivankravets, 2019-04-18:
Please re-test with upstream version https://docs.platformio.org/en/latest/platforms/ststm32.html#upstream

closed 2019-04-18 by ivankravets

muhammedtapis, 2021-03-20:
can you please share where the id script path is i cant find it in which file ?

commit e4a1454e (2019-04-18): Expand LDSCRIPT_PATH and fix issue when user name contains a space // Resolve #229