platformio/platform-ststm32 #231: bluepill_f103c8 won't blink with the arduino maple core

avaldebe, 2019-03-30:
The blink example will work fine for the default stm32duino core,
but it does nothing for the maple code.

Here is my config:

```
; stm32 core
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
build_flags = -D LED_BUILTIN=PC13

; maple core
[env:bluepill_maple]
platform = ststm32
board = bluepill_f103c8
framework = arduino
board_build.core = maple
build_flags = -D LED_BUILTIN=PC13

```

gmcn42, 2019-11-29:
I have the same problem. In fact the maple core seems to be completely broken, nothing compiled using it will work on the STM32F103C8T6. It will just lock up and do nothing.

By contrast, using the maple core (rogerclarkmelbourne repository) in the Arduino IDE works flawlessly with the same examples I tried, so the problem seems to be with Platform.io or our config (my ini looks the same as above).

pfeerick, 2020-03-15:
I suspect this was a mistake on our part. I was able to reproduce this just now with the below `platformio.ini` and code. I then changed the board type to `genericSTM32F103C8` (which also allows for `board_build.core = maple` to be removed), and it's working fine. 

Since the rogerclarkmelbourne/Arduino_STM32 (maple core) doesn't actually mention the blue pill in the boards.txt (and does have `Generic STM32F103C series`), but the stm32duino/Arduino_Core_STM32 does... I think we've been specifying the wrong board. This also makes sense given the default cores used. 🤦‍♂ 🤐 

```
[env:STM32]
platform = ststm32
board = bluepill_f103c8
board_build.core = maple
framework = arduino
upload_protocol = stlink
debug_tool = stlink
```

```
#include <Arduino.h>
void setup()
{
  pinMode(PC13 ,OUTPUT);
}

void loop()
{
  digitalWrite(PC13 ,HIGH);
  delay(500);
  digitalWrite(PC13 ,LOW);
  delay(500);
}
```

jtroyer76, 2020-03-27:
What I am finding is that if I have only: 

[env:STM32]
platform = ststm32
board = genericSTM32F103C8
framework = arduino

This build by default with the maple core... 

and if I specify anything other than maple for board type ti will default to the stm32duino/Arduino_Core_STM32  even as shown below.

[env:STM32]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
board_build.core = bla

pfeerick, 2020-03-28:
> and if I specify anything other than maple for board type

To be expected... since the test is only checking if maple is specified, else use the stm32duino core

https://github.com/platformio/platform-ststm32/blob/6f38950430250ea315bbee628b95e5ed680deee0/builder/frameworks/arduino.py#L35

jtroyer76, 2020-03-28:
That makes sense... just wanted to clarify that for the  generic board definition that maple core appears to be the default.

pfeerick, 2020-09-15:
> Is there any fix for this yet? I can't use Maple on bluepill at all, it
> compiles and upload but does nothing

Does it work with the platform.ini and code sample from my earlier comment?

https://github.com/platformio/platform-ststm32/issues/231#issuecomment-599163921

matias-levy, 2020-09-16:
> Does it work with the platform.ini and code sample from my earlier comment? [#231 (comment)](https://github.com/platformio/platform-ststm32/issues/231#issuecomment-599163921)
> […](#)

No, I am out of ideas. This works on the Arduino IDE