platformio/platform-ststm32 #232: Home: Could not initialize project

spam-receiver, 2019-04-04:
PIO Core Call Error: "The next files/directories have been created in C:\\Users\\User\\Documents\\PlatformIO\\Projects\\CKS32F103C8T6\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\n\nError: Processing genericSTM32F103C8 (platform: ststm32; board: genericSTM32F103C8; framework: libopencm3)\r\n\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103C8.html\r\nPLATFORM: ST STM32 > STM32F103C8 (20k RAM. 64k Flash)\r\nHARDWARE: STM32F103C8T6 72MHz 20KB RAM (64KB Flash)\r\nDEBUG: CURRENT(blackmagic) EXTERNAL(blackmagic, jlink, stlink)\r\nIOError: [Errno 2] No such file or directory: 'C:\\\\Users\\\\User\\\\.platformio\\\\packages\\\\framework-libopencm3\\\\lib\\\\Makefile':\r\nFile \"C:\\Users\\User\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\main.py\", line 169:\r\nenv.SConscript(\"$BUILD_SCRIPT\")\r\nFile \"C:\\Users\\User\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 541:\r\nreturn _SConscript(self.fs, *files, **subst_kw)\r\nFile \"C:\\Users\\User\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 250:\r\nexec _file_ in call_stack[-1].globals\r\nFile \"C:\\Users\\User\\.platformio\\platforms\\ststm32\\builder\\main.py\", line 90:\r\ntarget_elf = env.BuildProgram()\r\nFile \"C:\\Users\\User\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 224:\r\nreturn self.method(*nargs, **kwargs)\r\nFile \"C:\\Users\\User\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 107:\r\nenv.BuildFrameworks(env.get(\"PIOFRAMEWORK\"))\r\nFile \"C:\\Users\\User\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 224:\r\nreturn self.method(*nargs, **kwargs)\r\nFile \"C:\\Users\\User\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 319:\r\nSConscript(env.GetFrameworkScript(f), exports=\"env\")\r\nFile \"C:\\Users\\User\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 604:\r\nreturn method(*args, **kw)\r\nFile \"C:\\Users\\User\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 541:\r\nreturn _SConscript(self.fs, *files, **subst_kw)\r\nFile \"C:\\Users\\User\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 250:\r\nexec _file_ in call_stack[-1].globals\r\nFile \"C:\\Users\\User\\.platformio\\platforms\\ststm32\\builder\\frameworks\\libopencm3\\libopencm3.py\", line 181:\r\nget_source_files(root_dir)\r\nFile \"C:\\Users\\User\\.platformio\\platforms\\ststm32\\builder\\frameworks\\libopencm3\\libopencm3.py\", line 100:\r\nmkdata = parse_makefile_data(join(src_dir, \"Makefile\"))\r\nFile \"C:\\Users\\User\\.platformio\\platforms\\ststm32\\builder\\frameworks\\libopencm3\\libopencm3.py\", line 77:\r\nwith open(makefile) as f:\r\n [ERROR] Took 1.79 seconds"

valeros, 2019-06-28:
Thanks, fixed! Use the latest [upstream version](http://docs.platformio.org/en/latest/platforms/ststm32.html#stable-and-upstream-versions) of the platform to retest your project.

closed 2019-06-28 by valeros

spam-receiver, 2019-06-28:
> Thanks, fixed! Use the latest [upstream version](http://docs.platformio.org/en/latest/platforms/ststm32.html#stable-and-upstream-versions) of the platform to retest your project.

Thank you, now its work.