platformio/platform-ststm32 #234: Demo F030F4 default MCU startup file not found

jlmcgehee21, 2019-04-04:
```
[env:demo_f030f4]
platform = ststm32@5.3.0
board = demo_f030f4
framework = stm32cube
```

When using the above configuration, platformio fails to find the startup script for the default mcu listed in the `demo_f030f4` board configuration -- `STM32F030F4P6`.

I believe this is a failure of the creation of `search_path` here: https://github.com/platformio/platform-ststm32/blob/5a389c1383b95d8b0ef4cdcc0fbd3625888178e4/builder/frameworks/stm32cube.py#L77

I'm totally willing to submit a PR as a fix, but may need some guidance around what an ideal solution looks like. I could definitely fix _this_ issue, but don't want to break lookups for any other scenario.

valeros, 2019-04-04:
Hi @jlmcgehee21 ! Thanks for pointing it out. Looks like this mcu is a special case that requires a precise filename for the startup file. Anyway, it's been fixed in the dev branch. Would be great if you could test it:
```
[env:demo_f030f4]
platform = https://github.com/platformio/platform-ststm32.git
board = demo_f030f4
framework = stm32cube
```

valeros, 2019-05-09:
Feel free to reopen the issue if you still have any problems.

closed 2019-05-09 by valeros