platformio/platform-ststm32 #240: Add a new option which could generate .hex file in platformio.ini

Weeeendi, 2019-05-17:
I meet a problem that I can't upload code into my stm32 chip with UART ，so I want to find .hex file
upload by other tools. Unfortunately I can‘ t find it in my project，I hope you can creat a option in platformio.ini which could choose generate .hex when build phase.

valeros, 2019-05-20:
You can use an extra script option to generate the hex file, e.g.:
```ini
[env:nucleo_f401re]
platform = ststm32
framework = stm32cube
board = nucleo_f401re
extra_scripts = post:extra_script.py
```
and `extra_script.py`:
```python
Import("env")

# Custom HEX from ELF
env.AddPostAction(
    "$BUILD_DIR/${PROGNAME}.elf",
    env.VerboseAction(" ".join([
        "$OBJCOPY", "-O", "ihex", "-R", ".eeprom",
        "$BUILD_DIR/${PROGNAME}.elf", "$BUILD_DIR/${PROGNAME}.hex"
    ]), "Building $BUILD_DIR/${PROGNAME}.hex")
)
```

ivankravets, 2019-05-23:
There is an example in our docs => http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#before-pre-and-after-post-actions

closed 2019-05-23 by ivankravets