platformio/platform-ststm32 #247: BluePill board option with a CKS32F103C8T6 chip

exothink, 2019-06-13:
I have a bluepill board with an mcu made by CKS.  This chip is compatible the STM32f103 chip as per their website.

Using ST-link to load the chip throws an error because the CKS32F103C8T6 has a 'CPUTAPID 0x2ba01477' in openOCD.  The STM chip normally on these boards has a 'CPUTAPID 0x1ba01477'.

Please add a BluePill board with the CKS chip option.

PSIAlt, 2019-07-28:
Changing lines in file ~/.platformio/packages/tool-openocd/scripts/target/stm32f1x.cfg fixes the upload too:

```
      -set _CPUTAPID 0x1ba01477
      +set _CPUTAPID 0x2ba01477

```

Also, it would be great to be possible to pass `CPUTAPID` into openocd directly from platformio.init config.
And, user-friendly board option would be great!

akuhtz, 2019-10-01:
@PSIAlt @exothink  I had the same problem and found a solution here: https://www.reddit.com/r/AskElectronics/comments/bs0k3n/someone_heard_of_artery_mcus/eojs405/

```
upload_flags =
    -c
    set CPUTAPID 0x2ba01477
```

crafter999, 2019-10-12:
> @PSIAlt @exothink I had the same problem and found a solution here: https://www.reddit.com/r/AskElectronics/comments/bs0k3n/someone_heard_of_artery_mcus/eojs405/
> 
> ```
> upload_flags =
>     -c
>     set CPUTAPID 0x2ba01477
> ```

That worked for me too. I think it's because of the cheap Chinese clones.

dreamcat4, 2020-08-11:
This is the more compatible solution;

```diff
diff --git a/a/.platformio/packages/tool-openocd/scripts/target/stm32f1x.cfg b/b/.platformio/packages/tool-openocd/scripts/target/stm32f1x.cfg
index 3e85fb2..8ef0b2d 100644
--- a/a/.platformio/packages/tool-openocd/scripts/target/stm32f1x.cfg
+++ b/b/.platformio/packages/tool-openocd/scripts/target/stm32f1x.cfg
@@ -40,10 +40,15 @@ if { [info exists CPUTAPID] } {
    } {
       # this is the SW-DP tap id not the jtag tap id
       set _CPUTAPID 0x1ba01477
+
+      # simple 'replacement' workaround for chinese clone CS32F103C8T6 / CKS32F 103C8T6 NK1M5 1845 E
+      # however this disables the official ST chip as only 1 chip id can be specified this way
+      ; set _CPUTAPID 0x2ba01477
    }
 }
 
-swj_newdap $_CHIPNAME cpu -irlen 4 -ircapture 0x1 -irmask 0xf -expected-id $_CPUTAPID
+# workaround cross compatible with both official ST chip and chinese clone CS32F103C8T6 / CKS32F 103C8T6 NK1M5 1845 E
+swj_newdap $_CHIPNAME cpu -irlen 4 -ircapture 0x1 -irmask 0xf -expected-id $_CPUTAPID -expected-id 0x2ba01477
 dap create $_CHIPNAME.dap -chain-position $_CHIPNAME.cpu
 
 if {[using_jtag]} {
```

I believe it's worth it, because so many people end up with these Chinese clone boards.

Levi--G, 2020-08-25:
> This is the more compatible solution;
> I believe it's worth it, because so many people end up with these Chinese clone boards.

This would be great, it is the only way i found to get debugging working too. I am guessing since the upload flag is not used for debugging. Thanks for this solution! I hope this gets implemented in the next update.