platformio/platform-ststm32 #248: OpenOCD script not found after updated to 5.4.2

dogtopus, 2019-06-14:
```
GNU MCU Eclipse OpenOCD, 64-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-22-20:18)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
embedded:startup.tcl:56: Error: Can't find scripts/interface/stlink.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 56
```

ivankravets, 2019-06-14:
What is your board?

ivankravets, 2019-06-14:
Please update to the latest 5.4.3 https://github.com/platformio/platform-ststm32/releases/tag/v5.4.3

Thanks!

closed 2019-06-14 by ivankravets

SergeySn, 2019-08-10:
I have 5.4.3 but I have the same problem:

> Checking size .pioenvs/BIGTREE_SKR_PRO/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   4.5% (used 8780 bytes from 196608 bytes)
PROGRAM: [=         ]  13.9% (used 146108 bytes from 1048576 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, dfu, jlink, stlink
CURRENT: upload_protocol = stlink
Uploading .pioenvs/BIGTREE_SKR_PRO/firmware.elf
GNU MCU Eclipse OpenOCD, 64-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-22-20:18)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
embedded:startup.tcl:56: **Error: Can't find scripts/interface/stlink.cfg**
in procedure 'script' 
at file "embedded:startup.tcl", line 56
*** [upload] Error 1

ivankravets, 2019-08-11:
@SergeySn do you use the latest version of ST STM32 dev/platform? `pio update`?

SergeySn, 2019-08-11:
@ivankravets Thanks a lot for the quick reply.  If I use the latest, 5.5.0, than I get this error:
> ...platformio/packages/framework-arduinoststm32/cores/arduino/stm32/PinNames.h:16:12: error: 'PortA' undeclared here (not in a function)

It looks like there is already a bug open for this issue:
https://github.com/platformio/platform-ststm32/issues/267
But I don't see any solution there yet.

Stretto74, 2019-08-16:
Hi Guys, I'm just getting into the whole vscode & atom with platformio. ive managed to finallly get a build without errors on both platform vscode and atom but both are hitting me with the same as above error line scripts/interface/stlink.cfg error. 
I've tried to find a solution here and on other support pages specific to the bigtree skr pro board groups not having much luck. is it possible i can be pointed to a solution?

ivankravets, 2019-08-19:
@Stretto74 please update all packages via `pio update` and use the latest stable `ststm32` dev/platform in `platformio.ini`.