platformio/platform-ststm32 #259: workspace_dir not working in combination with "upload_protocol = dfu" and “upload_port = anything”

rwalli, 2019-07-12:
workspace_dir does not work in 4.0.0 if
```
upload_protocol = dfu
upload_port = anything
```
is set in platformio.ini as discussed [here](https://community.platformio.org/t/problem-with-workspace-dir/8629)

To reproduce:
```
    [platformio]
    workspace_dir = /tmp/pio-build/$PROJECT_HASH

    [env:bluepill_f103c8_128k]
    platform = ststm32
    framework = arduino
    board = bluepill_f103c8_128k
    upload_protocol = dfu
    upload_port = anything
```
and run “pio run -t upload”

as suggested replacing in "platformio/platform-ststm32/blob/develop/builder/main.py#L192":
`UPLOADCMD='$UPLOADER $UPLOADERFLAGS "$PROJECT_DIR/$SOURCES"'`
with
`UPLOADCMD='$UPLOADER $UPLOADERFLAGS "$SOURCES"'`
is fixing the issue.

valeros, 2019-07-17:
Hi @rwalli ! Please retest your project with the [upstream version of the platform](http://docs.platformio.org/en/latest/platforms/ststm32.html#stable-and-upstream-versions).

ivankravets, 2019-07-18:
Fixed in the latest version

closed 2019-07-18 by ivankravets