platformio/platform-ststm32 #261: Board request: OpenCM 9.04

zouden, 2019-07-18:
The OpenCM 9.04 from Robotis is a servo controller based on the Cortex M3 and the official reference board for controlling Dynamixel servos. An Arduino board definition already exists:

https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCM9.04/master/arduino/opencm_release/package_opencm9.04_index.json

Documentation about the board is available here:
 
http://emanual.robotis.com/docs/en/parts/controller/opencm904/

Support for this board in PlatformIO would be most excellent. The official library is already in the platformio repository (https://platformio.org/lib/show/6467/Dynamixel2Arduino/)

ruffson, 2019-11-24:
Very much in support of this! What would be needed? Board and platform definition?

AR17HY, 2022-01-15:
Does anyone have seen some work on this task ? I am trying to build code with VCS and PlatformIO for this board OpenCM9.04
I am currently using the generic STM32F103CB board and the library Dynamixel2Arduino without success...
Do you know if an alternative exist ?