platformio/platform-ststm32 #263: OpenOCD loads ELF file into Blue Pill, not BIN (for framework-arduinoststm32-maple), firmware is linked incorrectly

me21, 2019-07-18:
I'm building simple Blink example shipped with PlatformIO. I use slightly modified build environments:

```
[env:bluepill_f103c8]
platform = ststm32
framework = arduino
board = bluepill_f103c8
board_build.core = maple

[env:genericSTM32F103CB]
platform = ststm32
framework = arduino
board = genericSTM32F103CB
board_build.core = maple
```

If I upload using the first environment, the firmware doesn't work. If I upload using the second one, it does.

Reading the chip memory shows the ELF file header in the first case, that means OpenOCD has flashed ELF file, not converting it to BIN.

This must have something to do with the wrong linkage because the resulting firmware is incorrectly linked (corresponding `mem-flash.inc` specifies ROM origin as 0x8005000, while interrupt vector table address is 0x8000000 according to `platformio-build-stm32f1.py` script). Editing `mem-flash.inc` and setting ROM origin to 0x8000000 makes the first environment operational too.

valeros, 2019-07-19:
Hi @me21! For some reason, openocd writes additional headers to the beginning of the flash memory when an application has an offset in its linker script. Could you please try to add the next flag `-Wl,--nmagic` to `build_flags` field?

me21, 2019-07-19:
Looks like it now writes the firmware at 0x8005000 address, but since interrupt vector table address is 0x8000000, it doesn't work. But at least there's no ELF header.