platformio/platform-ststm32 #265: Redundant compilation flags for genericSTM32F103RC

saloid, 2019-07-19:
When i am trying to verbose build my project, I can see extra flags, which i think not necessary. Removing them (from C:\Users\Dmytro\.platformio\packages\framework-arduinoststm32-maple\tools\platformio-build-stm32f1.py) reduces firmware size.

`arm-none-eabi-g++ -o .pio\build\genericSTM32F103RC\src\main.cpp.o -c -std=gnu++11 -fno-rtti -fno-exceptions -Wall -mcpu=cortex-m3 -mthumb -ffunction-sections -fdata-sections -nostdlib -march=armv7-m --param max-inline-insns-single=500 -Wall -Os -DPLATFORMIO=40001 -DSTM32F103xE -DSTM32F1 -DARDUINO=10808 -DARDUINO_ARCH_STM32 -DARDUINO_ARCH_STM32F1 -DARDUINO_GENERIC_STM32F103R -DMCU_STM32F103RC -D__STM32F1__ -DBOARD_generic_stm32f103r -DF_CPU=72000000L -DVECT_TAB_ADDR=0x8000000 **-DERROR_LED_PORT=GPIOB -DERROR_LED_PIN=1 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1 -DSERIAL_USB -DGENERIC_BOOTLOADER** -Isrc -I.pio\libdeps\genericSTM32F103RC\TinyGSM_ID1287\src -IC:\users\dmytro\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple -IC:\users\dmytro\.platformio\packages\framework-arduinoststm32-maple\STM32F1\system\libmaple -IC:\users\dmytro\.platformio\packages\framework-arduinoststm32-maple\STM32F1\system\libmaple\include -IC:\users\dmytro\.platformio\packages\framework-arduinoststm32-maple\STM32F1\system\libmaple\stm32f1\include -IC:\users\dmytro\.platformio\packages\framework-arduinoststm32-maple\STM32F1\system\libmaple\usb\stm32f1 -IC:\users\dmytro\.platformio\packages\framework-arduinoststm32-maple\STM32F1\system\libmaple\usb\usb_lib -IC:\users\dmytro\.platformio\packages\framework-arduinoststm32-maple\STM32F1\variants\generic_stm32f103r src\main.cpp`

saloid, 2019-07-19:
As I understand, this is for debug LEDs, USB-Serial and, maybe, usb bootloader. But I am not using them all - is it possible to exclude them from final firmware if they are not needed (for example if upload_protocol set not in dfu)?

valeros, 2019-07-19:
Hi @saloid! Please take a look at this option http://docs.platformio.org/en/latest/projectconf/section_env_build.html#build-unflags

closed 2019-07-19 by valeros

saloid, 2019-07-20:
Yes, thanks, it almost works, but:
When I am trying to unflag options without '=' in it (for example -DSERIAL_USB) - this not works (this flag still presented while compiling). But `-DSERIAL_USB=1` unflags this option.
It's okay, but in that page example `build_unflags = -Os -std=gnu++11` somehow works: -Os disappears.