platformio/platform-ststm32 #266: OpenOCD script not found after updating to 5.4.3

nwright91, 2019-07-20:
GNU MCU Eclipse OpenOCD, 64-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-22-20:25)
Licensed under GNU GPL v2
For bug reports, read http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
embedded:startup.tcl:56: Error: Can't find scripts/interface/stlink.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 56

SergeySn, 2019-08-10:
I have the same problem:

> Checking size .pioenvs/BIGTREE_SKR_PRO/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   4.5% (used 8780 bytes from 196608 bytes)
PROGRAM: [=         ]  13.9% (used 146108 bytes from 1048576 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, dfu, jlink, stlink
CURRENT: upload_protocol = stlink
Uploading .pioenvs/BIGTREE_SKR_PRO/firmware.elf
GNU MCU Eclipse OpenOCD, 64-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-22-20:18)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
embedded:startup.tcl:56: Error: Can't find scripts/interface/stlink.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 56
*** [upload] Error 1

valeros, 2019-08-29:
Hi @nwright91 @SergeySn ! Have you tried the latest version of the `ststm32` platform (5.6.0)? Does it work?

SergeySn, 2019-08-29:
@valeros If I use 
`platform = ststm32@5.6.0`

Then it doesn't compile.  It gives me error:
`Compiling .pioenvs/BIGTREE_SKR_PRO/FrameworkArduino/stm32/analog.c.o
In file included from /home/serg/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/pinmap.h:22:0,
                 from /home/serg/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/PeripheralPins.h:34,
                 from /home/serg/.platformio/packages/framework-arduinoststm32/variants/BIGTREE_GENERIC_STM32F407_5X/variant.h:37,
                 from /home/serg/.platformio/packages/framework-arduinoststm32/system/STM32F4xx/stm32f4xx_hal_conf.h:4,
                 from /home/serg/.platformio/packages/framework-arduinoststm32/system/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal.h:30,
                 from /home/serg/.platformio/packages/framework-arduinoststm32/system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/stm32f4xx.h:250,
                 from /home/serg/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/stm32_def.h:35,
                 from /home/serg/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/PortNames.h:33,
                 from /home/serg/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/PortNames.c:30:
/home/serg/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/PinNames.h:16:12: error: 'PortA' undeclared here (not in a function)
   PA_0  = (PortA << 4) + 0x00,`

valeros, 2019-08-29:
@SergeySn Probably related to this #267

SergeySn, 2019-08-29:
@valeros Yes, they are related.  In #267 you said:
> Probably the best option is to roll back to the last working version of the platform.

But so far I was unable to find an older version of the platform where it all works.
If I roll back to ststm32@5.4.3 - it compiles, but when I try to upload it gives me:
> embedded:startup.tcl:56: Error: Can't find scripts/interface/stlink.cfg

And with ststm32@5.6.0 it gives me `error: 'PortA' undeclared` 

Any solution?

valeros, 2019-08-29:
As a temorary workaround, you can try to install the latest platform and then manually downgrade the framework version to `3.10500.0` in `platform.json` here `/home/serg/.platformio/platforms/ststm32/platform.json`
```
"framework-arduinoststm32": {
      "type": "framework",
      "optional": true,
      "version": "~3.10500.0"
}
```

Evg33, 2019-08-29:
https://github.com/MarlinFirmware/Marlin/issues/14672#issuecomment-521638243

SergeySn, 2019-08-29:
@valeros If I switch to `ststm32@5.6.0` and downgrade the framework version to 3.10500.0 in platform.json then it compiles, but when I try to upload, it gives me the same: 
`embedded:startup.tcl:56: Error: Can't find scripts/interface/stlink.cfg`

valeros, 2019-08-29:
@SergeySn What board do you use?

SergeySn, 2019-08-29:
@valeros I use BigTreeTech SKR Pro.

valeros, 2019-08-29:
@SergeySn There is no such board in our repository. Where can I find the manifest file for this board?

SergeySn, 2019-08-29:
What's a "manifest file"?

I can see `define BOARD_BIGTREE_SKR_PRO_V1_1    4207  // BigTreeTech SKR Pro v1.1 (STM32F407ZG)`
in
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/core/boards.h

valeros, 2019-08-29:
Could you try to open this file`buildroot/share/PlatformIO/boards/BigTree_SKR_Pro.json`  and replace its contents with the next config:
```
{
  "build": {
    "core": "stm32",
    "cpu": "cortex-m4",
    "extra_flags": "-DSTM32F407xx",
    "f_cpu": "168000000L",
    "hwids": [
      [
        "0x1EAF",
        "0x0003"
      ],
      [
        "0x0483",
        "0x3748"
      ]
    ],
    "ldscript": "stm32f407xg.ld",
    "mcu": "stm32f407zgt6",
    "variant": "BIGTREE_GENERIC_STM32F407_5X"
  },
  "debug": {
    "jlink_device": "STM32F407ZG",
    "openocd_target": "stm32f4x",
    "svd_path": "STM32F40x.svd"
  },
  "frameworks": [
    "arduino",
    "stm32cube"
  ],
  "name": "STM32F407ZG (192k RAM. 1024k Flash)",
  "upload": {
    "disable_flushing": false,
    "maximum_ram_size": 196608,
    "maximum_size": 1048576,
    "protocol": "stlink",
    "protocols": [
      "stlink",
      "dfu",
      "jlink"
    ],
    "require_upload_port": true,
    "use_1200bps_touch": false,
    "wait_for_upload_port": false
  },
  "url": "http://www.st.com/en/microcontrollers/stm32f407zg.html",
  "vendor": "Generic"
}

```

SergeySn, 2019-08-29:
@valeros Wow, with your BigTree_SKR_Pro.json file - it works.  Thanks a lot!

valeros, 2020-09-15:
Please reopen the issue if you still need any help.

closed 2020-09-15 by valeros