platformio/platform-ststm32 #267: Marlin2 does not compile with 5.5.0

Evg33, 2019-07-21:
Log:
```
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\LL\stm32yyxx_ll_ucpd.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\LL\stm32yyxx_ll_usart.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\LL\stm32yyxx_ll_usb.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\LL\stm32yyxx_ll_utils.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\PortNames.c.o
Compiling .pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\analog.c.o
In file included from C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/pinmap.h:22:0,
                 from C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PeripheralPins.h:34,
                 from C:\users\evg33\.platformio\packages\framework-arduinoststm32\variants\BIGTREE_GENERIC_STM32F407_5X/variant.h:35,
                 from C:\users\evg33\.platformio\packages\framework-arduinoststm32\system\STM32F4xx/stm32f4xx_hal_conf.h:4,
                 from C:\users\evg33\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_hal.h:30,
                 from C:\users\evg33\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F4xx\Include/stm32f4xx.h:250,
                 from C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\stm32_def.h:35,
                 from C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\PortNames.h:33,
                 from C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\PortNames.c:30:
C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:16:12: error: 'PortA' undeclared here (not in a function)
   PA_0  = (PortA << 4) + 0x00,
            ^~~~~
C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:33:12: error: 'PortB' undeclared here (not in a function); did you mean 'PortA'?
   PB_0  = (PortB << 4) + 0x00,
            ^~~~~
            PortA
C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:50:12: error: 'PortC' undeclared here (not in a function); did you mean 'PortB'?
   PC_0  = (PortC << 4) + 0x00,
            ^~~~~
            PortB
C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:68:12: error: 'PortD' undeclared here (not in a function); did you mean 'PortC'?
   PD_0  = (PortD << 4) + 0x00,
            ^~~~~
            PortC
C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:86:12: error: 'PortE' undeclared here (not in a function); did you mean 'PortD'?
   PE_0  = (PortE << 4) + 0x00,
            ^~~~~
            PortD
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\PortNames.c.o] Error 1
C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.c: In function 'get_adc_internal_channel':
C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.c:243:10: error: 'PADC_TEMP' undeclared (first use in this function); did you mean 'PADC_BASE'?
     case PADC_TEMP:
          ^~~~~~~~~
          PADC_BASE
C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.c:243:10: note: each undeclared identifier is reported only once for each function it appears 
in
C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.c:248:10: error: 'PADC_VREF' undeclared (first use in this function); did you mean 'PADC_TEMP'?
     case PADC_VREF:
          ^~~~~~~~~
          PADC_TEMP
C:\users\evg33\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.c:253:10: error: 'PADC_VBAT' undeclared (first use in this function); did you mean 'PADC_VREF'?
     case PADC_VBAT:
          ^~~~~~~~~
          PADC_VREF
*** [.pioenvs\BIGTREE_SKR_PRO\FrameworkArduino\stm32\analog.c.o] Error 1
============================================================================ [ERROR] Took 35.57 seconds ============================================================================

===================================================================================== [SUMMARY] ===================================================================================== 
Environment BIGTREE_SKR_PRO             [FAILED]
====================================================================== 1 failed, 0 succeeded in 35.67 seconds ======================================================================  
```

platformio.ini
```
[platformio]
src_dir      = Marlin
build_dir    = .pioenvs
lib_dir      = .piolib
libdeps_dir  = .piolibdeps
boards_dir   = buildroot/share/PlatformIO/boards
#env_default  = megaatmega2560
default_envs = BIGTREE_SKR_PRO

....

[env:BIGTREE_SKR_PRO]
#platform = ststm32@5.4.3
platform = ststm32
framework = arduino
board = BigTree_SKR_Pro
extra_scripts = pre:buildroot/share/PlatformIO/scripts/generic_create_variant.py
build_flags = ${common.build_flags}
  -DUSBCON -DUSBD_USE_CDC -DUSBD_VID=0x0483 -DUSB_PRODUCT=\"STM32F407ZG\"
  -DTARGET_STM32F4 -DSTM32F407_5ZX -DVECT_TAB_OFFSET=0x8000 -DHAVE_HWSERIAL6
lib_deps = ${common.lib_deps}
lib_ignore = Adafruit NeoPixel, c1921b4, TMC26XStepper, SailfishLCD, SailfishRGB_LED, SlowSoftI2CMaster
src_filter = ${common.default_src_filter} +<src/HAL/HAL_STM32>
monitor_speed = 250000
```

Evg33, 2019-07-21:
Broken at https://github.com/platformio/platform-ststm32/commit/3c27299f954a5e9b084db9e7e91c79cf9a1818aa

tpruvot, 2019-07-22:
so might be in https://github.com/stm32duino/Arduino_Core_STM32/commits/master ... i see the version was changed...

valeros, 2019-07-22:
Hi @Evg33 , @tpruvot ! In the 5.5.0 release, we switched to the latest STM32Duino (v1.6.1). I'm guessing the compilation fails because they changed the mechanism of how `stm32f4xx_hal_conf.h` is handled. Now there is common `stm32fxxx_hal_conf.h` per STM32 family (e.g. `system/STM32F4xx/stm32f4xx_hal_conf.h`)

SergeySn, 2019-08-11:
@valeros So, is there any solution or workaround?

valeros, 2019-08-12:
Probably the best option is to roll back to the last working version of the platform.

Evg33, 2019-08-15:
Is the problem in STM32Duino or in marlin ?

tpruvot, 2019-08-15:
STM32Duino yep cf https://github.com/MarlinFirmware/Marlin/issues/14672#issuecomment-521633437

SergeySn, 2019-08-29:
BTW, ststm32@5.6.0 it gives the same `error: 'PortA' undeclared`

Evg33, 2019-08-29:
@tpruvot delete tools section from file buildroot/share/PlatformIO/boards/BigTree_SKR_Pro.json
and compilation ok with new ststm32 :)
```
[env:BIGTREE_SKR_PRO]
#platform = ststm32@5.4.3
platform = ststm32

```

https://github.com/platformio/platform-ststm32/issues/266#issuecomment-526292630

Evg33, 2019-08-29:
@tpruvot delete tools section from buildroot/share/PlatformIO/boards/BigTree_SKR_Pro.json
and compilation ok with platform = ststm32

Thx 2 @valeros https://github.com/platformio/platform-ststm32/issues/266#issuecomment-526292630
```
[env:BIGTREE_SKR_PRO]
#platform = ststm32@5.4.3
platform = ststm32

```

Evg33, 2019-08-29:
https://github.com/MarlinFirmware/Marlin/issues/14672#issuecomment-526373765

closed 2019-08-29 by Evg33