platformio/platform-ststm32 #269: "sh: stm32flash: command not found" after fresh install of PIO

steffentrumpf, 2019-08-04:
While trying to upload to a "Generic STM32F103C8" via serial I get the following error: "sh: stm32flash: command not found".

It has been working flawlessly before. After messing around with an F4 core, it doesn't work anymore. I tried removing VS code and PIO including, what I think, all background files. After a fresh install I get the same error.

Any Idea why stm32flash is not installed/working?

OS: macOS 10.14.6
PIO Core: 4.0.0
VS code: 1.36.1

kinosang, 2019-08-06:
Same here

And after install stm32flash via Homebrew, it still not work.

ref #138

pfeerick, 2019-08-08:
Having the same issue on Linux, using the STM32F103C8/B and bluepill_f103c8/_128k board variants. A quick `which stm32flash` from the VSCode terminal doesn't find executable either, although that may be unrelated. The file is present in `
/home/pfeerick/.platformio/packages/tool-stm32duino/stm32flash/stm32flash` and will execute if run run manually. 

```
Platform ST STM32
--------
Checking ststm32                         @ 5.5.0          [Up-to-date]
Checking toolchain-gccarmnoneeabi        @ 1.70201.0      [Up-to-date]
Checking framework-arduinoststm32-maple  @ 1.10000.190719 [Up-to-date]
Checking tool-stm32duino                 @ 1.0.1          [Up-to-date]
Checking tool-openocd                    @ 2.1000.190422  [Up-to-date]
Checking tool-dfuutil                    @ 1.9.190708     [Up-to-date]
```

steffentrumpf, 2019-08-13:
Thanks a lot!
I did the things you mentioned and they worked.

sanyi, 2019-10-21:
I would add that on linux the error is:
```
Uploading .pio/build/bluepill_f103c8/firmware.bin
sh: 1: stm32flash: Permission denied
*** [upload] Error 127
```
somewhat misleading. But replacing `UPLOADER` to full path and removing the `basename` trimming of the `UPLOAD_PORT` works:
```
elif upload_protocol == "serial":
    env.Replace(
        UPLOADER="/home/atti/.platformio/packages/tool-stm32duino/stm32flash/stm32flash",
        UPLOADERFLAGS=[
            "-g", board.get("upload.offset_address", "0x08000000"),
            "-b", "115200", "-w"
        ],
        UPLOADCMD='$UPLOADER $UPLOADERFLAGS "$SOURCE" "$UPLOAD_PORT"'
    )
....
```

valeros, 2019-10-22:
Thanks for reporting! Should be fixed in the upstream version:
```
[env:upstream_develop]
platform = https://github.com/platformio/platform-ststm32.git
board = ...
```

commit 97dddaec (2019-10-22): Fix upload process via serial protocol // Resolve #269
closed 2019-10-22 by valeros