platformio/platform-ststm32 #270: Please add support for arduino-STM32L4.

darshanmirajkar, 2019-08-05:
Please add support for arduino-STM32L4.

See: https://github.com/GrumpyOldPizza/arduino-STM32L4, which supports the Butterfly and Ladybug

darshanmirajkar, 2019-08-06:
@valeros When can we get this any idea??

valeros, 2019-08-06:
Hi @darshanmirajkar! As soon as we have spare time for this feature. PlatformIO is an open-source project you can write build script for this framework by yourself and submit it as a pull request.

darshanmirajkar, 2019-08-06:
Tanks for replay @valeros. If you have any docs about build scripts please share me...ill also try

valeros, 2019-08-06:
Unfortunately, there is no documentation for this process, but you can find plenty of examples here https://github.com/platformio/platform-ststm32/tree/develop/builder/frameworks

darshanmirajkar, 2019-08-06:
@valeros Thank you... ill try.

darshanmirajkar, 2019-08-07:
@valeros I saw this tread [https://github.com/platformio/platform-ststm32/issues/28](here) buy you guys not added??

valeros, 2019-08-07:
We decided to switch all STM32 targets to the official [Arduino core](https://github.com/stm32duino/Arduino_Core_STM32)

darshanmirajkar, 2019-08-07:
ok.. but we are using libraries from [https://github.com/GrumpyOldPizza/arduino-STM32L4](this) core.

valeros, 2019-08-07:
Unfortunately, we don't have so much spare time to adapt all existing cores.