platformio/platform-ststm32 #271: Black f407ve board uploading failure with st-link v2

chjz1024, 2019-08-06:
I bought a black f407ve board and st-link v2 from a chinese store and would like to test programming it with platformio. When uploading the program the log seems to be okay but in fact my code didn't run as expected. I also tested debugging but the program always stucks at the very beginning and never went into `setup()` or `loop()` function. I tested the same code with arduino IDE and stm32duino library and nothing is wrong. I'm a bit confused because I think platformio uses stm32duino to support programming like arduino.

My platformio.ini file:

```ini
[env:black_f407ve]
platform = ststm32
board = black_f407ve
framework = arduino
```

main.cpp

```cpp
#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
  pinMode(PA1,OUTPUT);
  pinMode(PA5,OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(PA1, HIGH);
  digitalWrite(PA5, LOW);
  delay(1000);
  digitalWrite(PA1, LOW);
  digitalWrite(PA1, HIGH);
  delay(1000);
}
```

The log is as follows:

> \> Executing task: C:\Users\18245\.platformio\penv\Scripts\platformio.exe run --target upload <
> 
> 
> Processing black_f407ve (platform: ststm32; board: black_f407ve; framework: arduino)
> \--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> 
> Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/black_f407ve.html
> PLATFORM: ST STM32 5.5.0 > Black STM32F407VE
> HARDWARE: STM32F407VET6 168MHz, 192KB RAM, 512KB Flash
> DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
> PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32 3.10601.190716 (1.6.1), tool-stm32duino 1.0.1, tool-openocd 2.1000.190422 (10.0), tool-dfuutil 1.9.190708
> LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ soft
> Found 7 compatible libraries
> Scanning dependencies...
> No dependencies
> Compiling .pio\build\black_f407ve\FrameworkArduinoVariant\PeripheralPins.c.o
> Compiling .pio\build\black_f407ve\FrameworkArduinoVariant\variant.cpp.o
> Compiling .pio\build\black_f407ve\FrameworkArduino\HardwareSerial.cpp.o
> Compiling .pio\build\black_f407ve\FrameworkArduino\IPAddress.cpp.o
> Compiling .pio\build\black_f407ve\FrameworkArduino\Print.cpp.o
> Compiling .pio\build\black_f407ve\FrameworkArduino\RingBuffer.cpp.o
> Compiling .pio\build\black_f407ve\FrameworkArduino\Stream.cpp.o
> Compiling .pio\build\black_f407ve\FrameworkArduino\Tone.cpp.o
> Compiling .pio\build\black_f407ve\FrameworkArduino\USBSerial.cpp.o
> Compiling .pio\build\black_f407ve\FrameworkArduino\WInterrupts.cpp.o
> Compiling .pio\build\black_f407ve\FrameworkArduino\WMath.cpp.o
> Compiling .pio\build\black_f407ve\FrameworkArduino\WSerial.cpp.o
> Compiling .pio\build\black_f407ve\FrameworkArduino\WString.cpp.o
> Compiling .pio\build\black_f407ve\FrameworkArduino\abi.cpp.o
> Compiling .pio\build\black_f407ve\FrameworkArduino\avr\dtostrf.c.o
> Compiling .pio\build\black_f407ve\FrameworkArduino\board.c.o
> Compiling .pio\build\black_f407ve\FrameworkArduino\hooks.c.o
> Compiling .pio\build\black_f407ve\FrameworkArduino\itoa.c.o
[... 180 lines cut ...]
> Uploading .pio\build\black_f407ve\firmware.elf
> GNU MCU Eclipse OpenOCD, 64-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-22-20:25)
> Licensed under GNU GPL v2
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> debug_level: 1
> hla_swd
> adapter speed: 2000 kHz
> adapter_nsrst_delay: 100
> none separate
> adapter speed: 1800 kHz
> target halted due to debug-request, current mode: Thread 
> xPSR: 0x01000000 pc: 0x1fff3da0 msp: 0x20001d80
> adapter speed: 4000 kHz
> ** Programming Started **
> auto erase enabled
> wrote 16384 bytes from file .pio\build\black_f407ve\firmware.elf in 0.635992s (25.158 KiB/s)
> ** Programming Finished **
> ** Verify Started **
> verified 13236 bytes in 0.118000s (109.541 KiB/s)
> ** Verified OK **
> ** Resetting Target **
> adapter speed: 1800 kHz
> shutdown command invoked
> ============================================================================================== [SUCCESS] Took 23.42 seconds ==============================================================================================

valeros, 2019-08-06:
Hi @chjz1024! Looks like the firmware is broken. Does it work in Arduino IDE?

chjz1024, 2019-08-06:
Hi @valeros . Yes it works. That's why it confused me.

flair0626, 2019-08-07:
Processing demo_f030f4 (platform: ststm32; board: demo_f030f4; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/demo_f030f4.html
PLATFORM: ST STM32 5.5.0 > Demo F030F4
HARDWARE: STM32F030F4P6 48MHz, 4KB RAM, 16KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: tool-stm32duino 1.0.1, tool-dfuutil 1.9.190708, framework-arduinoststm32 3.10601.190716 (1.6.1), toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), tool-openocd 2.1000.190422 (10.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 7 compatible libraries
Scanning dependencies...
No dependencies
Checking size .pio\build\demo_f030f4\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [==        ]  17.9% (used 732 bytes from 4096 bytes)
PROGRAM: [=====     ]  50.9% (used 8332 bytes from 16384 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, jlink, serial, stlink
CURRENT: upload_protocol = stlink
Uploading .pio\build\demo_f030f4\firmware.elf
GNU MCU Eclipse OpenOCD, 64-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-22-20:25)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
hla_swd
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
adapter speed: 1000 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0x08001ea8 msp: 0x20001000
adapter speed: 8000 kHz
** Programming Started **
auto erase enabled
wrote 9216 bytes from file .pio\build\demo_f030f4\firmware.elf in 0.571309s (15.753 KiB/s)
** Programming Finished **
** Verify Started **
verified 8544 bytes in 0.066821s (124.867 KiB/s)
** Verified OK **
** Resetting Target **
adapter speed: 1000 kHz
shutdown command invoked

trylika, 2019-08-22:
I can confirm, having same issue with `black_f407ve` board target and exact same project (sample blink project) - no compilation/linker/upload errors, but binary is not executed.
Changing board target to `genericSTM32F407VET6` works, but ofc changes core used.
Compiling and Uploading works on Arduino IDE and binary works without any problems.
Directly uploading with `STM32CubeProgrammer` `.bin` generated in Arduino or with `genericSTM32F407VET6` target works, but with `black_f407ve` target it seems like binaries are not executed.

For reference some build logs, but in reality i cant see much difference except for:
`xPSR: 0x01000000 pc: 0x080001cc msp: 0x20020000` vs `xPSR: 0x01000000 pc: 0x08002f28 msp: 0x20030000`

for `black_f407ve`:
```
> Executing task in folder STM32Blink: platformio.exe run --target upload <

Processing stm32blink (platform: ststm32; board: black_f407ve; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/black_f407ve.html
PLATFORM: ST STM32 5.5.0 > Black STM32F407VE
HARDWARE: STM32F407VET6 168MHz, 192KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: tool-stm32duino 1.0.1, tool-dfuutil 1.9.190708, framework-arduinoststm32 3.10601.190716 (1.6.1), toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), tool-openocd 2.1000.190422 (10.0)       
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 24 compatible libraries
Scanning dependencies...
No dependencies
Checking size .pio\build\stm32blink\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   0.4% (used 828 bytes from 196608 bytes)
PROGRAM: [          ]   2.4% (used 12828 bytes from 524288 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, jlink, serial, stlink
CURRENT: upload_protocol = stlink
Uploading .pio\build\stm32blink\firmware.elf
GNU MCU Eclipse OpenOCD, 64-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-22-20:25)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
hla_swd
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
adapter speed: 1800 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x080001cc msp: 0x20020000
adapter speed: 4000 kHz
** Programming Started **
auto erase enabled
wrote 16384 bytes from file .pio\build\stm32blink\firmware.elf in 0.598720s (26.724 KiB/s)
** Programming Finished **
** Verify Started **
verified 13248 bytes in 0.097959s (132.071 KiB/s)
** Verified OK **
** Resetting Target **
adapter speed: 1800 kHz
shutdown command invoked
```

for `genericSTM32F407VET6`:
```
> Executing task in folder STM32Blink: platformio.exe run --target upload <

Processing stm32blink (platform: ststm32; board: genericSTM32F407VET6; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F407VET6.html
PLATFORM: ST STM32 5.5.0 > STM32F407VE (192k RAM. 512k Flash)
HARDWARE: STM32F407VET6 168MHz, 128KB RAM, 502.23KB Flash
DEBUG: Current (stlink) External (jlink, stlink)
PACKAGES: tool-stm32duino 1.0.1, tool-dfuutil 1.9.190708, toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0), tool-openocd 2.1000.190422 (10.0) 
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 21 compatible libraries
Scanning dependencies...
No dependencies
Checking size .pio\build\stm32blink\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=         ]   5.6% (used 7376 bytes from 131072 bytes)
PROGRAM: [          ]   3.8% (used 19620 bytes from 514288 bytes)
Configuring upload protocol...
AVAILABLE: dfu, jlink, stlink
CURRENT: upload_protocol = stlink
Uploading .pio\build\stm32blink\firmware.elf
GNU MCU Eclipse OpenOCD, 64-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-22-20:25)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
hla_swd
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
none separate
adapter speed: 1800 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08002f28 msp: 0x20030000
adapter speed: 4000 kHz
** Programming Started **
auto erase enabled
wrote 32768 bytes from file .pio\build\stm32blink\firmware.elf in 1.076965s (29.713 KiB/s)
** Programming Finished **
** Verify Started **
verified 19620 bytes in 0.157681s (121.512 KiB/s)
** Verified OK **
** Resetting Target **
adapter speed: 1800 kHz
shutdown command invoked
```

trylika, 2019-08-23:
Checked earlier versions, last version when `black_f407ve` target works properly is `5.4.3`, so this is broken only on version `5.5.0`

build log for reference:
```
> Executing task in folder STM32Blink: platformio.exe run --target upload <

Processing stm32blink (platform: ststm32@5.4.3; board: black_f407ve; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/black_f407ve.html
PLATFORM: ST STM32 5.4.3 > Black STM32F407VE
HARDWARE: STM32F407VET6 168MHz, 192KB RAM, 512KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: tool-stm32duino 1.0.1, framework-arduinoststm32 3.10500.190327 (1.5.0), toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), tool-openocd 2.1000.190422 (10.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 24 compatible libraries
Scanning dependencies...
No dependencies
Checking size .pio\build\stm32blink\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   0.5% (used 1080 bytes from 196608 bytes)
PROGRAM: [          ]   2.3% (used 12276 bytes from 524288 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, jlink, serial, stlink
CURRENT: upload_protocol = stlink
Uploading .pio\build\stm32blink\firmware.elf
GNU MCU Eclipse OpenOCD, 64-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-22-20:25)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
hla_swd
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
adapter speed: 1800 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x080023c8 msp: 0x20020000
adapter speed: 4000 kHz
** Programming Started **
auto erase enabled
wrote 16384 bytes from file .pio\build\stm32blink\firmware.elf in 0.631362s (25.342 KiB/s)
** Programming Finished **
** Verify Started **
verified 12696 bytes in 0.107823s (114.989 KiB/s)
** Verified OK **
** Resetting Target **
adapter speed: 1800 kHz
shutdown command invoked
```

trylika, 2019-08-23:
to be even more exact, found commit which broke the build: https://github.com/platformio/platform-ststm32/commit/3c27299f954a5e9b084db9e7e91c79cf9a1818aa

ktand, 2019-08-26:
@valeros, I believe the problem is caused by incorrect **maximum_ram_size** in the board configuration file: 

`"maximum_ram_size": 196608,`

should be:

`"maximum_ram_size": 131072,`

And this applies to **all STM32F4 boards**. The issue started to occur after the linker script was updated in the STM32 Arduino Core:

```
/* Highest address of the user mode stack */
_estack = 0x20000000 + LD_MAX_DATA_SIZE;    /* end of RAM */
```

which was previously:

`_estack = 0x20020000;    /* end of RAM */`

LD_MAX_DATA_SIZE is passed as maximum_ram_size and with a value of 196608 this exceeds the end of RAM.

ktand, 2019-08-26:
> Does it work when you change `maximum_ram_size` to `131072`

For me it does, yes.

valeros, 2019-08-26:
Nice catch @ktand! So the problem might be that we consider `CCRAM` memory in `maximum_ram_size` and Arduino core simply doesn't take it into account. 
Does it work when you change `maximum_ram_size` to `131072`

closed 2019-08-27 by valeros

trylika, 2019-08-27:
but should this change be done in `platform-ststm32` or in `STM32 Arduino Core` ?
because it looks like `maximum_ram_size` should be `196608`, it's just that `_estack` value should be 
```
0x20000000 + LD_MAX_DATA_SIZE - CCRAM_SIZE
```

ktand, 2019-08-27:
@valeros This applies to the **armed_v1** board as well as it is using STM32F407.

ktand, 2019-08-27:
@valeros I'm planning to submit a PR for Arduino Core STM32 to add LD_MAX_DATA_SIZE to the ldscript for armed_v1 (I'm the designer of the armed board).

valeros, 2019-08-27:
@ktand This is a temporary solution, as we haven't decided how to handle two memory banks of RAM. And since there is no `LD_MAX_DATA_SIZE` in ldscript for `armed_v1` board we prefer to leave it as it was before.

valeros, 2019-08-27:
Great, we'll update board manifest for `armed_v1` when a new version of the framework is released.

chjz1024, 2019-08-27:
@ktand @valeros 
Cool! After I change the `maximum_ram_size` to `131072` my code works. Thank you very much for your help.

commit ac83bd4e (2019-08-27): Fix maximum_ram_size for Arduino enabled boards (F4) by excluding CCM // Resolve #271