platformio/platform-ststm32 #272: arm semihosting support for ststm32 STM32F746G-DISCO / ST-Link

marcellodum, 2019-08-06:
I'm trying to get semihosting to work  on STM32F746G-DISCO. Current platform.ini configuration file is:

`platform = ststm32`
`board = disco_f746ng`
`framework = stm32cube`
`build_flags = -DF7 --specs=rdimon.specs -lrdimon -nostartfiles`
`; Enable Arm Semihosting in debugger to show logs.`
`debug_extra_cmds =` 
`    monitor arm semihosting enable`
`    set remotetimeout 100000`

The initialization code is:

`extern void initialise_monitor_handles(void);`
`initialise_monitor_handles();`
`printf("Hello World\n");`

Can't get any output on the debug console

valeros, 2019-08-06:
Hi @marcellodum! Because of `--specs=rdimon.specs` flag it's not that easy.
First,  `--specs=rdimon.specs` shouldn't be used with default `--specs=nosys.specs` so we need to disable it:
```
build_unflags =
    --specs=nosys.specs 
    -lnosys
```
Also, `--specs=rdimon.specs` must be added directly to the linker and unfortunately, that's only possible by using an extra script:
`extra_script.py:`
```
Import("env")

# Add semihosting feature
env.Append(
    LINKFLAGS=["--specs=rdimon.specs"],
    LIBS="rdimon"
)
```
The result is something like this:
```
platform = ststm32
board = disco_f746ng
framework = stm32cube
build_flags = -DF7
extra_scripts = extra_script.py
build_unflags =
    --specs=nosys.specs 
    -lnosys
debug_extra_cmds =
    monitor arm semihosting enable
```

marcellodum, 2019-08-06:
Hey @valeros :) Thanks for the quick reply. I did exactly how you described (and makes total sense) still can't get any output on the debug console. Here is a sample from the verbose compilation and linking maybe it could help:

`arm-none-eabi-gcc -o .pio/build/disco_f746ng/src/main.o -c -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m7 -nostdlib -DPLATFORMIO=40000 -DSTM32F746xx -DF7 -DUSE_HAL_DRIVER -DF_CPU=216000000L -Iinclude -Isrc -Ilib/diag -I/Users/marcelloscacchetti/.platformio/packages/framework-stm32cube/f7/Drivers/CMSIS/Include -I/Users/marcelloscacchetti/.platformio/packages/framework-stm32cube/f7/Drivers/CMSIS/Device/ST/STM32F7xx/Include -I/Users/marcelloscacchetti/.platformio/packages/framework-stm32cube/f7/Drivers/STM32F7xx_HAL_Driver/Inc -I/Users/marcelloscacchetti/.platformio/packages/framework-stm32cube/f7/Drivers/BSP/Components/Common -I/Users/marcelloscacchetti/.platformio/packages/framework-stm32cube/f7/Drivers/BSP/STM32746G-Discovery src/main.c`

`...`

`arm-none-eabi-gcc -o .pio/build/disco_f746ng/firmware.elf -T /Users/marcelloscacchetti/.platformio/packages/framework-stm32cube/platformio/ldscripts/STM32F746NG_DEFAULT.ld -Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m7 --specs=nano.specs --specs=rdimon.specs .pio/build/disco_f746ng/src/main.o -L/Users/marcelloscacchetti/.platformio/platforms/ststm32/ldscripts -L.pio/build/disco_f746ng -L/Users/marcelloscacchetti/.platformio/packages/framework-stm32cube/f7/Drivers/CMSIS/Lib/GCC -L/Users/marcelloscacchetti/.platformio/packages/framework-stm32cube/platformio/ldscripts -Wl,--start-group .pio/build/disco_f746ng/libcd3/libdiag.a -lc -lgcc -lm -lstdc++ .pio/build/disco_f746ng/libFrameworkBSP.a .pio/build/disco_f746ng/libFrameworkHALDriver.a .pio/build/disco_f746ng/libFrameworkCMSISDevice.a -lrdimon -Wl,--end-group`

Thanks!

valeros, 2019-08-06:
I think it works, it's just not visible in the debug console as it appears on the stdout of the OpenOCD process. It seems we cannot overcome this limitation properly at the moment (at least I'm not aware of any possible solution). 
But here is a workaround, you just need to start the OpenOCD server in a separate window:
navigate to OpenOCD folder from your terminal:
`cd /Users/marcelloscacchetti/.platformio/packages/tool-openocd`
Run the next command:
`bin/openocd.exe -f scripts/interface/stlink.cfg  -f scripts/target/stm32f7x.cfg`

and specify the debug tool as `custom` in `platformio.ini`:
```
debug_tool = custom
debug_port = :3333
```

marcellodum, 2019-08-06:
@valeros works like a charm! You are a rockstar :)

closed 2019-10-22 by valeros

Crazor, 2020-05-16:
Works great! This needs better integration, though.

For the time being, here's a task to run openocd in a terminal window inside VSCode (paste into the project's `.vscode/tasks.json`):

```
{
    "version": "2.0.0",
    "tasks": [
        {
            "type": "shell",
            "command": "cd ~/.platformio/packages/tool-openocd && bin/openocd -f scripts/interface/stlink.cfg -f target/stm32f4x.cfg",
            "args": [
            ],
            "problemMatcher": [
                "$platformio"
            ],
            "label": "OpenOCD (semihosting output)"
        }
    ]
}
```

Run this task before starting the debug session. It creates a terminal window (you need to switch to it manually to see the semihosting outout).

I wasn't able to get `~` to resolve correctly inside the args, so I had to hack all into the command.