platformio/platform-ststm32 #274: Home: Could not install platform

pandcy, 2019-08-17:
PIO Core Call Error: "PlatformManager: Installing ststm32\r\nststm32 @ 5.5.0 has been successfully installed!\r\nPackageManager: Installing toolchain-gccarmnoneeabi @ >=1.60301.0,<1.80000.0\r\n\n\nError: Detected unknown package 'toolchain-gccarmnoneeabi'"

valeros, 2020-08-13:
At the moment only GCC v9.3.1 is available for ARM platforms, please try the following configuration:

```
[env:override_default_toolchain]
platform = ststm32
board = ...
framework = ...
platform_packages =
  toolchain-gccarmnoneeabi@~1.90301.0
```

hacker-cb, 2022-12-21:
@ivankravets hi!

May be it's better to specify <= 1.90301.0 here: https://github.com/platformio/platform-ststm32/blob/76958b476e042f578e0f6a6a74b8d47ca767db22/platform.json#L76

ivankravets, 2022-12-21:
@valeros, can we switch to the latest `10.3.1` as default? https://registry.platformio.org/tools/platformio/toolchain-gccarmnoneeabi ?

romanjeff, 2024-04-20:
Hi... the version numbering thing seems to be messing up the platform's ability to install properly and generate projects. The "version" constraint in the platform.json for toolchain-gccarmnoneeabi appears to be preventing "optionalVersions" from being checked, even though the most up to date versions are listed under optionalVersions. In order to get a working project generation I had to remove the version line entirely, then it worked. Maybe better to just list known compatible versions than rely on the equality comparisons that are out of date with the actual version numbering?