platformio/platform-ststm32 #275: Board Support: BigTree SKR Pro V1.1

skypuppy, 2019-08-17:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [x] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

pfeerick, 2019-09-24:
You can create a folder in your `$HOMEDIR/.platform` directory called `boards` and put the custom  [BigTree_SKR_Pro.json](https://raw.githubusercontent.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/master/firmware/Marlin-2.0.x-SKR-Pro/buildroot/share/PlatformIO/boards/BigTree_SKR_Pro.json) in it.

jaygrovr, 2019-09-28:
pfeerick, that is not working for me either.. can bigtree just update their firmware and make it work..i dont want to be spending time trying to get something to compile.

pfeerick, 2019-09-28:
When you say 

> that is not working for me either

What isn't working? What is the error message? How is it going to be fixed without even knowing what errors are coming up?

Following the instructions I have above - which requires you to do TWO things - copy the board and ensure the contents of variaents has the two BigTree variants folders, just compiled fine. If it actually works on their device is another things, as I don't have one. 

[image]

1) The board configuration file:
[image]

2) The variants folders from `firmware/Marlin-2.0.x-SKR-Pro/buildroot/share/PlatformIO/variants/` (I copied them into `framework-arduinoststm32`, and then was created when I did the first build as it seems a specific version of the framework was needed, but maybe platformio copied the variants folders as well? - I didn't need to copy them manually is all I can say. Try copying them into the `framework-arduinoststm32@3.10500.190327/variants` folder as well if not already present and it's still not working`.

[image]

[image]

LinoBarreca, 2019-11-05:
what doesn't work exactly?
I can build without problems for that board.