platformio/platform-ststm32 #277: AttributeError: 'NoneType' object has no attribute 'endswith'

yurybotov, 2019-08-17:
PIO Core Call Error: "The current working directory /home/yury/Documents/PlatformIO/Projects/testIskraNetduino will be used for the project.\n\nThe next files/directories have been created in /home/yury/Documents/PlatformIO/Projects/testIskraNetduino\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing netduino2plus (platform: ststm32; board: netduino2plus; framework: arduino)\n--------------------------------------------------------------------------------\nPackageManager: Installing framework-arduinoststm32-maple @ ~1.10000.190516\nframework-arduinoststm32-maple @ 1.10000.190719 has been successfully installed!\nCorePackageManager: Installing tool-scons @ ~2.20501.7\ntool-scons @ 2.20501.7 has been successfully installed!\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/ststm32/netduino2plus.html\nPLATFORM: ST STM32 5.5.0 > N2+\nHARDWARE: STM32F405RGT6 168MHz, 192KB RAM, 1MB Flash\nDEBUG: Current (stlink) External (jlink, stlink)\nPACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190719 (1.0.0)\nWarning! Cannot find linker script for the current target!\n\nLDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf\nLDF Modes: Finder ~ chain, Compatibility ~ soft\nFound 4 compatible libraries\nScanning dependencies...\nNo dependencies\nAttributeError: 'NoneType' object has no attribute 'endswith':\n  File \"/home/yury/.platformio/penv/local/lib/python2.7/site-packages/platformio/builder/main.py\", line 126:\n    env.SConscript(\"$BUILD_SCRIPT\")\n  File \"/home/yury/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py\", line 541:\n    return _SConscript(self.fs, *files, **subst_kw)\n  File \"/home/yury/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py\", line 250:\n    exec _file_ in call_stack[-1].globals\n  File \"/home/yury/.platformio/platforms/ststm32/builder/main.py\", line 217:\n    \"bin\", \"dfu-suffix\"),\n  File \"/home/yury/.platformio/penv/lib/python2.7/posixpath.py\", line 70:\n    elif path == '' or path.endswith('/'):\n========================== [ERROR] Took 13.70 seconds =========================="

valeros, 2019-08-19:
Hi @yurybotov ! Thanks, fixed in the latest development branch. Please try the next config:
```
[env:netduino2plus]
platform = https://github.com/platformio/platform-ststm32.git
board = netduino2plus
framework = arduino
```

valeros, 2019-08-27:
Should be fixed in the latest release `5.6.0`. Feel free to reopen if you still need help.

closed 2019-08-27 by valeros