platformio/platform-ststm32 #281: Can not compile Stm32f103c8  core: STM32duino

pasindu-sandima, 2019-09-02:
My platform.ini configuration
```

[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
board_build.core = STM32Duino
```

Error I get while compiling

```
Processing genericSTM32F103C8 (platform: ststm32; board: genericSTM32F103C8; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------ 
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103C8.html
PLATFORM: ST STM32 5.6.0 > STM32F103C8 (20k RAM. 64k Flash)
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: framework-arduinoststm32 3.10601.190716 (1.6.1), toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <VL53L0X> 1.0.2
|   |-- <Wire> 1.0
Linking .pio\build\genericSTM32F103C8\firmware.elf
arm-none-eabi-g++: error: Sandima\.platformio\packages\framework-arduinoststm32\variants\PILL_F103XX\ldscript.ld: No such file or directory
*** [.pio\build\genericSTM32F103C8\firmware.elf] Error 1

```

valeros, 2019-09-02:
There are only two officially supported cores:
`stm32` - https://github.com/stm32duino/Arduino_Core_STM32
`maple` - https://github.com/rogerclarkmelbourne/Arduino_STM32

closed 2019-09-02 by valeros

pasindu-sandima, 2019-09-03:
Yes. STM32duino means stm32. There is no issue with that. Can you please help me with the issue I have

saloid, 2020-07-18:
Would be nice to enumerate both of this options on [this page](https://docs.platformio.org/en/latest/platforms/ststm32.html).
Because there is unclear which option should I use to explicitly define stm32duino core.