platformio/platform-ststm32 #282: Can not compile Stm32f103c8 core: STM32duino

pasindu-sandima, 2019-09-03:
@valeros I changed my platform.ini file as you said. But it did not solve my issue. I get the same issure as #281 .
Glad if you can help

pasindu-sandima, 2019-09-03:
My platform.ini file

```
[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
board_build.core = stm32
```
Terminal Output

```
Processing genericSTM32F103C8 (platform: ststm32; board: genericSTM32F103C8; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103C8.html
PLATFORM: ST STM32 5.6.0 > STM32F103C8 (20k RAM. 64k Flash)
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: framework-arduinoststm32 3.10601.190716 (1.6.1), toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...     
No dependencies
Compiling .pio\build\genericSTM32F103C8\FrameworkArduinoVariant\PeripheralPins.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduinoVariant\variant.cpp.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\Print.cpp.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\RingBuffer.cpp.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\USBSerial.cpp.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\WInterrupts.cpp.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\WSerial.cpp.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\WString.cpp.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\abi.cpp.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\board.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\hooks.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\itoa.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\main.cpp.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\new.cpp.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\pins_arduino.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_adc.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_adc_ex.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_can.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cec.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_comp.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_comp_ex.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cortex.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_crc.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_crc_ex.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cryp.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cryp_ex.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dac.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dac_ex.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dcmi.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dcmi_ex.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dfsdm.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dfsdm_ex.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dma.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dma2d.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dma_ex.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_dsi.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_eth.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_eth_ex.c.o
[... 124 lines cut ...]
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\timer.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\twi.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\uart.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\usb\cdc\cdc_queue.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\usb\cdc\usbd_cdc.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\usb\cdc\usbd_cdc_if.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\usb\hid\usbd_hid_composite.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\usb\hid\usbd_hid_composite_if.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\usb\usb_device_core.c.o
[... cut ...]
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\wiring_pulse.cpp.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\wiring_shift.c.o
Compiling .pio\build\genericSTM32F103C8\FrameworkArduino\wiring_time.c.o
Compiling .pio\build\genericSTM32F103C8\src\main.cpp.o
Linking .pio\build\genericSTM32F103C8\firmware.elf
arm-none-eabi-g++: error: Sandima\.platformio\packages\framework-arduinoststm32\variants\PILL_F103XX\ldscript.ld: No such file or directory
*** [.pio\build\genericSTM32F103C8\firmware.elf] Error 1
```

pasindu-sandima, 2019-09-03:
main.cpp
```
#include <Arduino.h>
// #include <VL53L0X.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

valeros, 2019-09-03:
@pasindu-sandima  Do you have a whitespace character in your user name?

pasindu-sandima, 2019-09-03:
No.

valeros, 2019-09-03:
Please compile your project in verbose mode an attach here the build log. (Just open the terminal in IDE and run `pio run -v`)

pasindu-sandima, 2019-09-03:
```
L_F103XX" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\variants\PILL_F103XX" "C:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\HAL\stm32yyxx_hal_tim.c"
arm-none-eabi-gcc -o .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_tim_ex.c.o -c -std=gnu11 -Os -mcpu=cortex-m3 -mthumb -ffunction-sections -fdata-sections -Wall -nostdlib --param max-inline-insns-single=500 -DPLATFORMIO=40003 -DSTM32F103xB -DSTM32F1 -DSTM32F1xx -DARDUINO=10808 -DARDUINO_ARCH_STM32 -DARDUINO_GENERICSTM32F103C8 -DBOARD_NAME=\"GENERICSTM32F103C8\" -DHAL_UART_MODULE_ENABLED "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\cores\arduino\avr" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\LL" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb\hid" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb\cdc" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F1xx_HAL_Driver\Inc" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F1xx_HAL_Driver\Src" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\system\STM32F1xx" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\system\Middlewares\ST\STM32_USB_Device_Library\Core\Inc" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\system\Middlewares\ST\STM32_USB_Device_Library\Core\Src" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\CMSIS\CMSIS\Core\Include" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F1xx\Include" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F1xx\Source\Templates\gcc" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\cores\arduino" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\variants\PILL_F103XX" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\variants\PILL_F103XX" "C:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\HAL\stm32yyxx_hal_tim_ex.c"
arm-none-eabi-gcc -o .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_timebase_rtc_alarm_template.c.o -c -std=gnu11 -Os -mcpu=cortex-m3 -mthumb -ffunction-sections -fdata-sections -Wall -nostdlib --param max-inline-insns-single=500 -DPLATFORMIO=40003 -DSTM32F103xB -DSTM32F1 -DSTM32F1xx -DARDUINO=10808 -DARDUINO_ARCH_STM32 -DARDUINO_GENERICSTM32F103C8 -DBOARD_NAME=\"GENERICSTM32F103C8\" -DHAL_UART_MODULE_ENABLED "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\cores\arduino\avr" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\LL" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb\hid" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb\cdc" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F1xx_HAL_Driver\Inc" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F1xx_HAL_Driver\Src" "-IC:\msys64\home\Pasindu 
Sandima\.platformio\packages\framework-arduinoststm32\system\STM32F1xx" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\system\Middlewares\ST\STM32_USB_Device_Library\Core\Inc" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\system\Middlewares\ST\STM32_USB_Device_Library\Core\Src" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\CMSIS\CMSIS\Core\Include" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F1xx\Include" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F1xx\Source\Templates\gcc" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\cores\arduino" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\variants\PILL_F103XX" "-IC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\variants\PILL_F103XX" "C:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\HAL\stm32yyxx_hal_timebase_rtc_alarm_template.c"
arm-none-eabi-gcc -o .pio\build\genericSTM32F103C8\FrameworkArduino\stm32\HAL\stm32yyxx_hal_timebase_tim_template.c.o -c -std=gnu11 -Os -mcpu=cortex-m3 -mthumb -ffunction-sections -fdata-sections -Wall -nostdlib --param max-inline-insns-single=500 -DPLATFORMIO=40003 -DSTM32F103xB -DSTM32F1 -DSTM [... cut ...] ricSTM32F103C8 "-LC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\variants\PILL_F103XX" "-LC:\msys64\home\Pasindu Sandima\.platformio\packages\framework-arduinoststm32\CMSIS\CMSIS\DSP\Lib\GCC" -Wl,--start-group -larm_cortexM3l_math -lc -lm -lgcc -lstdc++ -Wl,--end-group
arm-none-eabi-g++: error: Sandima\.platformio\packages\framework-arduinoststm32\variants\PILL_F103XX\ldscript.ld: No such file or directory
*** [.pio\build\genericSTM32F103C8\firmware.elf] Error 1
```

pasindu-sandima, 2019-09-03:
The ld file is there in the directory
when i go through file explorer

fpistm, 2019-09-03:
@valeros asked you if you have a whitespace in your "**_Windows_**" user name and it seems yes.

Look  this line:
`arm-none-eabi-g++: error: Sandima\.platformio\packages\framework-arduinoststm32\variants\PILL_F103XX\ldscript.ld: No such file or directory`

it misses `C:\msys64\home\Pasindu `

pasindu-sandima, 2019-09-03:
Can it be changed?

pasindu-sandima, 2019-09-03:
It worked. Thanks a lot

closed 2019-09-03 by pasindu-sandima