platformio/platform-ststm32 #283: multiple definition of `premain()`, `main()` when using MySensors

pfeerick, 2019-09-09:
When using the MySensors library, linking fails with...

```
.pio/build/genericSTM32F103CB/src/main.cpp.o: In function `premain()':
main.cpp:(.text.startup._Z7premainv+0x0): multiple definition of `premain()'
.pio/build/genericSTM32F103CB/FrameworkArduino/main.cpp.o:main.cpp:(.text.startup._Z7premainv+0x0): first defined here
.pio/build/genericSTM32F103CB/src/main.cpp.o: In function `main':
main.cpp:(.text.startup.main+0x0): multiple definition of `main'
.pio/build/genericSTM32F103CB/FrameworkArduino/main.cpp.o:main.cpp:(.text.startup.main+0x0): first defined here
collect2: error: ld returned 1 exit status
```
... most likely due to the library also having a `premain()` and `main()` in the hal folder. On the Arduino IDE, using the rogerclarkmelbourne/Arduino_STM32 core it works just fine, but with 

```
[env:genericSTM32F103CB]
platform = ststm32
framework = arduino
board = genericSTM32F103CB
board_build.core = maple
lib_deps = 
  548 ; MySensors
```
and the LightSensor example (picked as one of the shorter ones which should 'just compile/work' for any board)  - with the requisite `#include <Arduino.h>` addition - it fails to build with PlatformIO. adding `lib_archive = no` does not make any noticeable difference. The same code **will** compile successfully for an atmelavr/uno target with PlatformIO.

 <details>
 <summary>Click to expand - Arduino IDE Compile verbose log</summary>

```
/usr/share/arduino/arduino-builder -dump-prefs -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/pfeerick/.arduino15/packages -hardware /home/pfeerick/Arduino/hardware -tools /usr/share/arduino/tools-builder -tools /home/pfeerick/.arduino15/packages -libraries /home/pfeerick/Arduino/libraries -fqbn=Arduino_STM32:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=DFUUploadMethod,cpu_speed=speed_72mhz,opt=osstd -ide-version=10809 -build-path /tmp/arduino_build_191534 -warnings=none -build-cache /tmp/arduino_cache_557302 -prefs=build.warn_data_percentage=75 -verbose /tmp/arduino_modified_sketch_736219/BatteryPoweredSensor.ino
/usr/share/arduino/arduino-builder -compile -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/pfeerick/.arduino15/packages -hardware /home/pfeerick/Arduino/hardware -tools /usr/share/arduino/tools-builder -tools /home/pfeerick/.arduino15/packages -libraries /home/pfeerick/Arduino/libraries -fqbn=Arduino_STM32:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=DFUUploadMethod,cpu_speed=speed_72mhz,opt=osstd -ide-version=10809 -build-path /tmp/arduino_build_191534 -warnings=none -build-cache /tmp/arduino_cache_557302 -prefs=build.warn_data_percentage=75 -verbose /tmp/arduino_modified_sketch_736219/BatteryPoweredSensor.ino
Using board 'genericSTM32F103C' from platform in folder: /home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1
Using core 'maple' from platform in folder: /home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1
Detecting libraries used...
/home/pfeerick/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8002000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10809 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple/include -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple/stm32f1/include -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple/usb/stm32f1 -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple/usb/usb_lib -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/cores/maple -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/variants/generic_stm32f103c /tmp/arduino_build_191534/sketch/BatteryPoweredSensor.ino.cpp -o /dev/null
/home/pfeerick/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8002000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10809 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple/include -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple/stm32f1/include -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple/usb/stm32f1 -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple/usb/usb_lib -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/cores/maple -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/variants/generic_stm32f103c -I/home/pfeerick/Arduino/libraries/MySensors /tmp/arduino_build_191534/sketch/BatteryPoweredSensor.ino.cpp -o /dev/null
/home/pfeerick/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8002000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10809 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple/include -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple/stm32f1/include -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple/usb/stm32f1 -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple/usb/usb_lib -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/cores/maple -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/variants/generic_stm32f103c -I/home/pfeerick/Arduino/libraries/MySensors -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/libraries/EEPROM /tmp/arduino_build_191534/sketch/BatteryPoweredSensor.ino.cpp -o /dev/null
/home/pfeerick/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8002000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10809 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple/include -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple/stm32f1/include -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple/usb/stm32f1 -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/system/libmaple/usb/usb_lib -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/cores/maple -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/variants/generic_stm32f103c -I/home/pfeerick/Arduino/libraries/MySensors -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/libraries/EEPROM -I/home/pfeerick/Arduino/hardware/Arduino_STM32/STM32F1/libraries/SPI/src /tmp/arduino_build_191534/sketch/BatteryPoweredSensor.ino.c [... cut ...] aple/usb/usb_lib -I/home/pfeerick/.platformio/packages/framework-arduinoststm32-maple/STM32F1/variants/generic_stm32f103c /home/pfeerick/.platformio/packages/framework-arduinoststm32-maple/STM32F1/cores/maple/libmaple/usb/stm32f1/usb_reg_map.c
arm-none-eabi-gcc -o .pio/build/genericSTM32F103CB/FrameworkArduino/libmaple/usb/usb_lib/usb_core.c.o -c -std=gnu11 -mcpu=cortex-m3 -mthumb -ffunction-sections -fdata-sections -nostdlib -march=armv7-m --param max-inline-insns-single=500 -Wall -Os -DPLATFOR

mtiutiu, 2019-09-20:
@pfeerick One way to solve this using the advanced features of platformio is like this:

In `platformio.ini` in the `env` section of your project add this line:
`extra_scripts = pre:fix_main.py`

Then create the `fix_main.py` file in the same directory as `platformio.ini` containing this:

```
Import("env")
import re
import os

STM32_FRAMEWORK_DIR = env['PROJECTPACKAGES_DIR'] + "/framework-arduinoststm32-maple"

def sed(file, pattern, replacement):
  with open(file, "r") as sources:
    lines = sources.readlines()
  with open(file, "w") as sources:
    for line in lines:
      sources.write(re.sub(pattern, replacement, line))

print("\n================================= STM32 FRAMEWORK PATCHER =================================\n")
if os.path.isdir(STM32_FRAMEWORK_DIR):
  for root, dirs, files in os.walk(STM32_FRAMEWORK_DIR):
    for file in files:
      if 'main.cpp' in file:
        main_cpp_file = os.path.join(root, file)
        print("[INFO] Patching file: {}".format(main_cpp_file))
        sed(main_cpp_file, '.*int main', '__attribute__ ((weak)) int main')
        sed(main_cpp_file, '.*void premain', '__attribute__(( constructor (101))) __attribute__ ((weak)) void premain') 
else:
  print("[WARN] Couldn't find stm32 framework directory! Searched for: {}".format(STM32_FRAMEWORK_DIR))
print("\n============================================================================================\n")
```

This doesn't solve the initial problem of course but until then it's just a quick workaround let's say. What it does in essence is before building the project it will search for all the `main.cpp` files in the arduino stm32 framework directory that platformio fetches and does a regex search and replace for lines that start with `int main` adding the `__attribute__ ((weak))` in front. Same for `void premain`.
Maybe the above can be done in a more elegant way but this is what I've come up so far and it seems to work. It also prints some information if it succeeded or not.
  Now this is not 100% foolproof as it relies on some predefined strings to search and replace which may change in the future.

Hope it helps for now.

pfeerick, 2019-09-21:
Sounds nice! Thanks for that! :) Handy `sed` / match 'n replace function there too! :)

tiegs, 2020-07-31:
For anyone struggling with the same issue:
"PROJECTPACKAGES_DIR" was changed to "PROJECT_PACKAGES_DIR".

If you get "KeyError: 'PROJECTPACKAGES_DIR':" change the expression accordingly.

fpistm, 2020-07-31:
This issue should be solved since the 1.8.0 release of the STM32 core as I've planned to support MySensors and I've met this issue so I've fix it.

tiegs, 2020-08-04:
I'm using platformio 4.3.4 and ST STM32 8.0.0, hitting the following error (building MySensors for BlackPill):

`
Building in release mode
Linking .pio/build/genericSTM32F103C8/firmware.elf
/home/tiegs/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/genericSTM32F103C8/src/main.cpp.o: in function `premain()':
main.cpp:(.text.startup._Z7premainv+0x0): multiple definition of `premain()'; .pio/build/genericSTM32F103C8/FrameworkArduino/main.cpp.o:main.cpp:(.text.startup._Z7premainv+0x0): first defined here
/home/tiegs/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/genericSTM32F103C8/src/main.cpp.o: in function `main':
main.cpp:(.text.startup.main+0x0): multiple definition of `main'; .pio/build/genericSTM32F103C8/FrameworkArduino/main.cpp.o:main.cpp:(.text.startup.main+0x0): first defined here
collect2: error: ld returned 1 exit status
*** [.pio/build/genericSTM32F103C8/firmware.elf] Error 1
`

I'm a little confused by "1.8.0 relase". Does this refer to 8.0 or is there another libary involved?

fpistm, 2020-08-04:
I'm talking about those version:
https://github.com/stm32duino/Arduino_Core_STM32/releases

For PIO, V7.0.0 include the STM32 core version 1.9.0:
https://github.com/platformio/platform-ststm32/tree/v7.0.0

So as I mentioned the MySensors library requires some update to handle properly this core like it has been done for Roger's core.

pfeerick, 2020-08-05:
> I'm a little confused by "1.8.0 relase".

Have a closer look at the build output - it will tell you up the top (1) the platformio package version and (2) the framework version. PlatformIO package 7.0.0 onwards has used arduinoSTM32 1.9.0. 

[image]

@fpistm Not quite sure what you mean here ...
> This issue should be solved since the 1.8.0 release of the STM32 core

... since the issue was related to the maple/rogerclark BSP/core. Do you mean you were hoping to get MySensors to add support for the STSTM32 BSP/core after changes made in 1.8.0? i.e. like seeing [this PR](https://github.com/mysensors/MySensors/pull/1422) merged?

fpistm, 2020-08-05:
Oh sorry, my mistake I thought it was with STM32 core not Roger's. Maybe a regression.
So, it seems strange it does not compile, last time I've tested it compile with Arduino. Did you test with Arduino ?

About the PR, I was not aware there is already one. That's fine as it seems what I've made allows to port properly the library. I will give a try.

tiegs, 2020-08-06:
Thanks for the clarification. I'm using arduinoststm32 v 1.9

```
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103C8.html
PLATFORM: ST STM32 8.0.0 > STM32F103C8 (20k RAM. 64k Flash)
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 4.10900.200602 (1.9.0) 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
```

Building the PR mentioned above I still hit the "multiple definition of `main'" issue:
```
/home/teggers/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/genericSTM32F103C8/src/main.cpp.o: in function `premain()':
main.cpp:(.text.startup._Z7premainv+0x0): multiple definition of `premain()'; .pio/build/genericSTM32F103C8/FrameworkArduino/main.cpp.o:main.cpp:(.text.startup._Z7premainv+0x0): first defined here
/home/teggers/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/genericSTM32F103C8/src/main.cpp.o: in function `main':
main.cpp:(.text.startup.main+0x0): multiple definition of `main'; .pio/build/genericSTM32F103C8/FrameworkArduino/main.cpp.o:main.cpp:(.text.startup.main+0x0): first defined here
collect2: error: ld returned 1 exit status
*** [.pio/build/genericSTM32F103C8/firmware.elf] Error 1
```

platformio.ini
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
;board_build.core = maple
framework = arduino

upload_protocol = serial
upload_port = /dev/ttyUSB0

monitor_speed = 115200

build_flags = -Os

lib_deps =
    Ethernet2@1.0.4
    ;MySensors@2.3.2
    ;https://github.com/mysensors/MySensors.git#development # 2.4 is WAY smaller than 2.3.2
    https://github.com/KooLru/MySensors.git#stm32_cores # MyS Branch for ststm32 build core (instead of maple)
```

pfeerick, 2020-08-07:
> Oh sorry, my mistake I thought it was with STM32 core not Roger's. Maybe a regression.

No problem ...  to err is human and all that jazz ;)

> So, it seems strange it does not compile, last time I've tested it compile with Arduino. Did you test with Arduino ?

When I initially had this issue, yes. As I said in the initial opening for the issue: "On the Arduino IDE, using the rogerclarkmelbourne/Arduino_STM32 core it works just fine." - hence the issue being lodged against PlatformIO since it's something PIO is doing differenet to the Arduino IDE, when using Rogers BSP. I haven't touched this since, so it's time for a revisit and poke and prod to see if anything has changes now that things have been updated a bit... :monocle_face: Plus I didn't know about the alternate 'STM32' BSP library forks, so that is a step forward! :smile_cat:

Gobol, 2020-10-19:
Hard'n'dirty workaround for STM32F1 boards:

Force __attribute__((weak)) before framework-arduino preinit() and main() functions... 

Change main.cpp in **~.platformio/packages/framework-arduinoststm32/cores/arduino** with :

`/*
  main.cpp - Main loop for Arduino sketches
  Copyright (c) 2005-2013 Arduino Team.  All right reserved.

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
*/

#define ARDUINO_MAIN
#include "Arduino.h"

// Force init to be called *first*, i.e. before static object allocation.
// Otherwise, statically allocated objects that need HAL may fail.
__attribute__((constructor(101))) __attribute__ ((weak)) void premain()
{

  // Required by FreeRTOS, see http://www.freertos.org/RTOS-Cortex-M3-M4.html
#ifdef NVIC_PRIORITYGROUP_4
  HAL_NVIC_SetPriorityGrouping(NVIC_PRIORITYGROUP_4);
#endif
#if (__CORTEX_M == 0x07U)
  // Defined in CMSIS core_cm7.h
#ifndef I_CACHE_DISABLED
  SCB_EnableICache();
#endif
#ifndef D_CACHE_DISABLED
  SCB_EnableDCache();
#endif
#endif

  init();
}

/*
 * \brief Main entry point of Arduino application
 */
__attribute__ ((weak))
int main(void)
{
  initVariant();

  setup();

  for (;;) {
#if defined(CORE_CALLBACK)
    CoreCallback();
#endif
    loop();
    serialEventRun();
  }

  return 0;
}
`

**build-upload-log:**
`Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8.html
PLATFORM: ST STM32 (8.1.0) > BluePill F103C8
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 4.10900.200819 (1.9.0) 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - tool-dfuutil 1.9.200310 
 - tool-openocd 2.1000.200630 (10.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <MySensors> 2.4.0-alpha+sha.df9c00a
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|   |-- <EEPROM> 2.0.1
|   |-- <IWatchdog> 1.0.0
Building in release mode
Checking size .pio/build/bluepill_f103c8/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  13.8% (used 2832 bytes from 20480 bytes)
Flash: [====      ]  42.4% (used 27776 bytes from 65536 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, dfu, jlink, mbed, stlink
CURRENT: upload_protocol = stlink
Uploading .pio/build/bluepill_f103c8/firmware.elf`

**platformio.ini:**
 
`[env:bluepill_f103c8]
; extra_scripts = pre:fix_main.py
platform = ststm32
board = bluepill_f103c8
;board_build.core = maple
framework = arduino

upload_protocol = stlink
debug_port = stlink
debug_tool = stlink
monitor_port = /dev/cu.usbmodem14103
monitor_speed = 115200
monitor_filters = time, default

lib_deps = 
    https://github.com/KooLru/MySensors.git#stm32_cores`