platformio/platform-ststm32 #286: Support for STM32F767 Development board (including the SDRAM)

resetcoder, 2019-09-16:
[http://www.chinalctech.com/cpzx/STM32_Study_Board/354.html](http://www.chinalctech.com/cpzx/STM32_Study_Board/354.html)

[https://www.banggood.com/STM32F767-Development-Board-Cortex-M7-Small-System-Board-STM32F767IGT6-STM32-p-1420406.html?cur_warehouse=CN](https://www.banggood.com/STM32F767-Development-Board-Cortex-M7-Small-System-Board-STM32F767IGT6-STM32-p-1420406.html?cur_warehouse=CN)

Unfortunately I can't find any pdf about the board yet.

uzi18, 2019-09-16:
@valeros can't find if this platform supports https://github.com/stm32duino/Arduino_Core_STM32 ?

it has https://github.com/stm32duino/Arduino_Core_STM32/tree/master/variants/NUCLEO_F767ZI support

resetcoder, 2019-09-16:
> @valeros can't find if this platform supports https://github.com/stm32duino/Arduino_Core_STM32 ?
> 
> it has https://github.com/stm32duino/Arduino_Core_STM32/tree/master/variants/NUCLEO_F767ZI support already

I'am already aware that Nucleo 767 is supported, but the board I'am asking for has additional 8MB SDRAM and 16MB Flash. The MCU has built-in external ram support, I don't really know if the compiler can use it at all, or how it is used. This is why I requested support, maybe someone knows about it.

I ordered the board lately, when it arrives I try my best to make it work with PlatformIO, then I post the results, but it would be great if someone has any experience with it.

ParetoYXE, 2020-08-18:
Did anything ever come of this?