platformio/platform-ststm32 #287: CMSIS-DAP support for Generic STM32 Boards

hez2010, 2019-09-19:
I modified `genericSTM32F407VET6.json` manually and append "cmsis-dap" in `protocols` section.
[image]

And I edit platformio.ini:
```ini
upload_protocol = cmsis-dap
debug_tool = cmsis-dap
```

Then I build and successfully upload the built firmware to my board.
[image]

I think you can add the cmsis-dap support for Generic STM32 boards.

My platformio.ini:
```ini
[env:genericSTM32F407VET6]
platform = ststm32
board = genericSTM32F407VET6
framework = cmsis
test_transport = custom
debug_tool = cmsis-dap
upload_protocol = cmsis-dap
lib_extra_dirs = .
build_flags = 
    -fexceptions
    -DUSE_STDPERIPH_DRIVER
```

jxltom, 2020-07-01:
Hey thanks for sharing.

Can you make the debug with cmsis-dap working?

jxltom, 2020-07-01:
For me, error is ```Error: Unknown debug tool `cmsis-dap`. Please use one of `jlink, stlink` or `custom````

hez2010, 2020-07-01:
It worked fine previously, but I have no idea about how it's going now since currently I'm not working with IoT development.

jxltom, 2020-07-01:
Ok, thanks for reply anyway.

gitneko, 2020-09-30:
The documentation to CMSIS-DAP debugging is quite misleading. I'm trying to use `cmsis-dap` as `debug_tool` as instructed by the documentation (<https://docs.platformio.org/en/v5.0.1/plus/debug-tools/cmsis-dap.html>) but it leads to a python error when trying to rebuild intelisense index:
```
DebugInvalidOptionsError: Unknown debug tool `cmsis-dap`. Please use one of `atmel-ice, blackmagic, jlink` or `custom`:
```

Fresh PlatformIO installation with Atmel SAM platform and Arduino MKR1000 board.

Core: v5.0.1
Home: v3.3.0

Is there something that I need to have installed to make this work?

424778940z, 2020-11-12:
> For me, error is `` Error: Unknown debug tool `cmsis-dap`. Please use one of `jlink, stlink` or `custom ``

you guys all missed the important part lol

"I modified genericSTM32F407VET6.json manually and append "cmsis-dap" in protocols section."

You HAVE TO edit the board define file.

commit 252752d7 (2020-11-26): Allow using CMSIS-DAP as upload protocol // Resolve #287
closed 2020-11-26 by valeros