platformio/platform-ststm32 #289: Debug breakpoint is not followed at MBED

SuperBoss9, 2019-09-24:
Hello everyone,

I found out that PIO doesn't stop at MBED breakpoint. For example, I use NUCLEO STM32F103RB if I set up a breakpoint then Debugger doesn't stop at the brekapoint. Instead of that it stops at the beginning of main function.
Then I can debug more or less well but if I continue execution of the program via "play" then debugger doesn't stop at the breakpoint again.

PS. Mbed Studio works well with its debugger and it can stop at my breakpoints.

valeros, 2019-09-24:
That's strange, breakpoint at main function is only temporary, you can disable it by specifying the following:
```
[env:debug]
platform = ...
board = ...
debug_init_break =
```

SuperBoss9, 2019-09-25:
Didn't help. Debugging started from main. But then after re-run it stopped at my breakpoint.
[image]

First stop at a "reset handler" then at the first string of main then at my breakpoint.

Test the same with Arduino. Absolutely equal behaviour.

It seems that it work at this machine.

closed 2019-09-25 by SuperBoss9