platformio/platform-ststm32 #290: framework-stm32cube: Custom HSE_VALUE is "overridden" by value from package's template stm32f4xx_hal_conf

mirao, 2019-09-25:
**Steps to reproduce:**

- Download the project https://github.com/mirao/stm32f4discovery/tree/master/stm32cube-printfOverUSB_UART
It's a demo of "printf" over UART on STM32F4 Discovery. It's using an external PL2303 converter to be able to print from MCU to PC's serial console
- Comment out the "build_flags" lines in `platform.ini`, this way:
```ini
; build_flags =
;     -DHSE_VALUE=8000000U
```
- Follow the README.md (how to connect device, build and upload) 
- When app is built and uploaded to MCU, open serial console and see output

**Expected result:**
- text "Hello World!"

**Actual result:**
- displayed characters are corrupted:
[image]

The problem is that the constant HSE_VALUE originally set to **8MHz** by STM32Cube (see `#define` in `Inc/stm32f4xx_hal_conf.h` in project's root) isn't taken into account. App uses the default value from `.platformio/packages/framework-stm32cube/f4/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_conf.h` instead and it is **25Mhz**. 

But serial/input output over UART **doesn't work** well with such frequency.

**Workaround:**
- Hardcode the constant value to `platformi.ini`. It takes precedence before `#define` in source code
```ini
build_flags =
    -DHSE_VALUE=8000000U
``` 
**Used SW and HW:**
- MCU STM32F407-DISC1
- Ubuntu 19.04
- VSCode 1.38.1
- pio 4.0.3
- ststm32 5.6.0
- framework-stm32cube 2.0.181130
- STM32CubeMX 5.3.0
- platformio.platformio-ide@1.8.3

valeros, 2019-09-26:
Your workaround with `build_flags` is the best solution at the moment. We plan to improve the build script for `stm32cube` framework, so it'll be able to handle such cases.

mirao, 2019-09-27:
Just FYI:
I've found out, that the external HSE oscillator was enabled because I confirmed the dialog _Initialize all peripherals with their default mode?_ in STM32CubeMX when I was creating a new project.
But the demo works even with internal HSI. It can be enabled in STM32CubeMX in "Clock Configuration => PLL Source HSI". HSI has only one frequency 16MHz and thus there's no need to apply any workaround.

raphaelmor, 2019-12-10:
Just adding my 2 cents (I spent a few days struggling with the same issue with I2S).
From what I understand (being a newbie with STM32, HAL, and Platformio), stm32f4xx_hal_conf.h is supposed to be copied into the application code and customized. I have no idea how to do that (given that platformio is so good at hiding those details), so I used the workaround of using build flags too.