platformio/platform-ststm32 #294: examples/mbed-custom-target does not compile

hbaqueiro, 2019-10-03:
```
baqueiro@baqueiro-Avell:mbed-custom-target (develop)$ pio run
Processing custom_nucleo_l452re_mbed (platform: ststm32; framework: mbed; board: custom_nucleo_l452re_mbed)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/custom_nucleo_l452re_mbed.html
PLATFORM: ST STM32 5.6.0 > ST Nucleo L452RE
HARDWARE: STM32L452RET6 80MHz, 64KB RAM, 256KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, jlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-mbed 5.51304.190826 (5.13.4)
Collecting mbed sources...
Detected custom target file
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 4 compatible libraries
Scanning dependencies...
No dependencies
Linking .pio/build/custom_nucleo_l452re_mbed/firmware.elf
.pio/build/custom_nucleo_l452re_mbed/src/TARGET_STM32L452xE/device/TOOLCHAIN_GCC_ARM/startup_stm32l452xx.o: In function `LoopFillZerobss':
/home/baqueiro/Desktop/platform-ststm32/examples/mbed-custom-target/src/TARGET_STM32L452xE/device/TOOLCHAIN_GCC_ARM/startup_stm32l452xx.s:110: undefined reference to `SystemInit'
collect2: error: ld returned 1 exit status
*** [.pio/build/custom_nucleo_l452re_mbed/firmware.elf] Error 1
========================================================================================================= [FAILED] Took 4.14 seconds =========================================================================================================

```

valeros, 2019-10-04:
The example has been updated according to the mbed v5.14.0 but it's not released yet. You can use [this revision](https://github.com/platformio/platform-ststm32/tree/afad63896082350aa964717c03acf677d38446b5/examples/mbed-custom-target).

hbaqueiro, 2019-10-04:
The revision does not compile either...
`baqueiro@baqueiro-Avell:mbed-custom-target (develop)$ pio run
Processing custom_nucleo_l452re_mbed (platform: ststm32; framework: mbed; board: custom_nucleo_l452re_mbed)
-----------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/custom_nucleo_l452re_mbed.html
PLATFORM: ST STM32 5.6.0 > ST Nucleo L452RE
HARDWARE: STM32L452RET6 80MHz, 64KB RAM, 256KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, jlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-mbed 5.51304.190826 (5.13.4)
Collecting mbed sources...
Detected custom target file
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 4 compatible libraries
Scanning dependencies...
No dependencies
Linking .pio/build/custom_nucleo_l452re_mbed/firmware.elf
.pio/build/custom_nucleo_l452re_mbed/src/TARGET_STM32L452xE/device/TOOLCHAIN_GCC_ARM/startup_stm32l452xx.o: In function `LoopFillZerobss':
/home/baqueiro/Desktop/platform-ststm32/examples/mbed-custom-target/src/TARGET_STM32L452xE/device/TOOLCHAIN_GCC_ARM/startup_stm32l452xx.s:110: undefined reference to `SystemInit'
collect2: error: ld returned 1 exit status
*** [.pio/build/custom_nucleo_l452re_mbed/firmware.elf] Error 1
=============================================================== [FAILED] Took 4.44 seconds ===============================================================`

valeros, 2019-10-04:
Please upload the whole build log to https://pastebin.com. In that revision the `SystemInit` function is defined [here](https://github.com/platformio/platform-ststm32/blob/afad63896082350aa964717c03acf677d38446b5/examples/mbed-custom-target/src/TARGET_STM32L452xE/TARGET_NUCLEO_L452RE/system_clock.c#L71).

valeros, 2020-08-03:
Closing the issue as stale. Feel free to reopen it or create a new issue with your feedback.

closed 2020-08-03 by valeros