platformio/platform-ststm32 #295: Unknown upload protocol olimex-arm-usb-ocd-h

NicHub, 2019-10-05:
I am trying to use my Olimex ARM-USB-OCD-H debugger for firmware uploading, but PlatformIO returns the following error:

    Warning! Unknown upload protocol olimex-arm-usb-ocd-h

I followed the instructions found here:

https://docs.platformio.org/en/latest/plus/debug-tools/olimex-arm-usb-ocd-h.html

The board is a STM32-E407 and here is my platformio.ini:

```ini
[platformio]
default_envs = black_f407zg

[env]
monitor_speed = 115200
build_flags =
    -D BAUD_RATE=${env.monitor_speed}
    -D HSE_VALUE=12000000
    -D ENABLE_HWSERIAL3 ; Serial on BOOT header
    -D ENABLE_HWSERIAL6 ; Serial on UEXT connector
    -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC ; Serial USB (USB CDC)
	-D USBCON
	-D USBD_VID=0x0483
	-D USB_MANUFACTURER="OLIMEX"
	-D USB_PRODUCT="\"STM32_E-407_board\""
	-D HAL_PCD_MODULE_ENABLED

[env:black_f407zg]
platform = ststm32
board = black_f407zg
framework = arduino
build_flags = ${env.build_flags}
upload_protocol = olimex-arm-usb-ocd-h
```

Is there something to install to make it work?

pfeerick, 2019-10-07:
I don't know about installing something, but the current reason for that error is because the stm32 platform is not supported using this debugger with PlatformIO, as shown by the 'Platforms' section of the instructions / info page you linked. 

https://aptly.io/en/blog/openocd-stm32

NicHub, 2019-10-07:
Thank you for your reply. Now I know that even if a debugger is supported by PlatformIO that doesn’t mean it is true for all platforms...

Is it on the roadmap to support the STM32 platform with this debugger ?

Thank you for the link. I’ll check it later.

valeros, 2019-10-14:
Hi @NicHub ! Is there a reason why you didn't try custom settings? For example:
```
[platformio]
default_envs = black_f407zg

[env]
monitor_speed = 115200
build_flags =
    -D BAUD_RATE=${env.monitor_speed}
    -D HSE_VALUE=12000000
    -D ENABLE_HWSERIAL3 ; Serial on BOOT header
    -D ENABLE_HWSERIAL6 ; Serial on UEXT connector
    -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC ; Serial USB (USB CDC)
	-D USBCON
	-D USBD_VID=0x0483
	-D USB_MANUFACTURER="OLIMEX"
	-D USB_PRODUCT="\"STM32_E-407_board\""
	-D HAL_PCD_MODULE_ENABLED

[openocd]
flags = 
    -s
    $PROJECT_PACKAGES_DIR/tool-openocd/scripts
    -f
    interface/ftdi/olimex-arm-usb-ocd-h.cfg
    -f
    target/stm32f4x.cfg

[env:black_f407zg]
platform = ststm32
board = black_f407zg
framework = arduino
build_flags = ${env.build_flags}

debug_tool = custom
upload_flags =
    ${openocd.flags}
    -c
    program {$SOURCE} verify reset; shutdown;
upload_command = $PROJECT_PACKAGES_DIR/tool-openocd/bin/openocd $UPLOAD_FLAGS

debug_server =
    $PLATFORMIO_CORE_DIR/packages/tool-openocd/bin/openocd
    ${openocd.flags}
```

NicHub, 2019-10-14:
Thank you for your reply. I didn’t try custom settings because I had no clue how to configure them. I tried your `platformio.ini`, but it fails with the following message:

    sh: /tool-openocd/bin/openocd: No such file or directory

Is there something I need to install? Note that I am working on macOS Mojave.

valeros, 2019-10-14:
Hm, try this one:
```
upload_command = $PLATFORMIO_CORE_DIR/packages/tool-openocd/bin/openocd $UPLOAD_FLAGS
```

NicHub, 2019-10-14:
Unfortunately it doesn’t work:

    sh: /packages/tool-openocd/bin/openocd: No such file or directory

valeros, 2019-10-14:
Hmm, try to specify the full path to the openocd binary, it's located somewhere here `~/.platformio/packages/tool-openocd`, so the upload command should look like this:

`upload_command = /Users/USER/.platformio/packages/tool-openocd/bin/openocd $UPLOAD_FLAGS`

NicHub, 2019-10-14:
I tried 

    upload_command = ~/.platformio/packages/tool-openocd/bin/openocd $UPLOAD_FLAGS

> Note that `~` seems to be OK in PlatformIO.

But it doesn’t work still this time the error is longer. I pressed the reset button before uploading and tried different positions for the Boot Mode Selection jumpers, but the result is always the same.

    Uploading .pio/build/black_f407zg/firmware.elf
    xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-15:21)
    Licensed under GNU GPL v2
    For bug reports, read
            http://openocd.org/doc/doxygen/bugs.html
    Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
    Info : clock speed 2000 kHz
    Info : JTAG tap: stm32f4x.cpu tap/device found: 0x4ba00477 (mfg: 0x23b (ARM Ltd.), part: 0xba00, ver: 0x4)
    Info : JTAG tap: stm32f4x.bs tap/device found: 0x06413041 (mfg: 0x020 (STMicroelectronics), part: 0x6413, ver: 0x0)
    Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
    Info : Listening on port 3333 for gdb connections
    Info : JTAG tap: stm32f4x.cpu tap/device found: 0x4ba00477 (mfg: 0x23b (ARM Ltd.), part: 0xba00, ver: 0x4)
    Info : JTAG tap: stm32f4x.bs tap/device found: 0x06413041 (mfg: 0x020 (STMicroelectronics), part: 0x6413, ver: 0x0)
    target halted due to debug-request, current mode: Thread 
    xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
    ** Programming Started **
    Info : device id = 0x10076413
    Info : flash size = 1024 kbytes
    Error: Invalid ACK (7) in DAP response
    Error: Failed to write memory at 0x20001370
    Error: error writing to flash at address 0x08000000 at offset 0x00000000
    embedded:startup.tcl:449: Error: ** Programming Failed **
    in procedure 'program' 
    in procedure 'program_error' called at file "embedded:startup.tcl", line 508
    at file "embedded:startup.tcl", line 449
    *** [upload] Error 1
    ============================================================== [FAILED] Took 16.93 seconds ==============================================================
    The terminal process terminated with exit code: 1

valeros, 2019-10-14:
Are you sure the wiring is correct? Also please make sure JTAG pins aren't used by the current firmware.

NicHub, 2019-10-15:
The wiring must be correct because I use a STM32-E407 board from Olimex with a 20-pin connector that is connected directly to the JTAG programmer. So there is no cabling of my own. 

I flashed an empty firmware via USB and then a simple blink firmware via the JTAG debugger and now the upload seems to work.

I also tried the debugger, it starts correctly (success at the end), but never stops on breakpoints.

Here is my `platformio.ini` file content:

```ini
[platformio]
default_envs = black_f407zg

[env]
monitor_speed = 115200
build_flags =
    -D BAUD_RATE=${env.monitor_speed}
    -D HSE_VALUE=12000000
    -D ENABLE_HWSERIAL3 ; Serial on BOOT header
    -D ENABLE_HWSERIAL6 ; Serial on UEXT connector
    -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC ; Serial USB (USB CDC)
	-D USBCON
	-D USBD_VID=0x0483
	-D USB_MANUFACTURER="OLIMEX"
	-D USB_PRODUCT="\"STM32_E-407_board\""
	-D HAL_PCD_MODULE_ENABLED

[openocd]
flags =
    -s
    $PROJECT_PACKAGES_DIR/tool-openocd/scripts
    -f
    interface/ftdi/olimex-arm-usb-ocd-h.cfg
    -f
    target/stm32f4x.cfg

[env:black_f407zg]
platform = ststm32
board = black_f407zg
framework = arduino
build_flags = ${env.build_flags}

debug_tool = custom
upload_flags =
    ${openocd.flags}
    -c
    program {$SOURCE} verify reset; shutdown;
upload_command = ~/.platformio/packages/tool-openocd/bin/openocd $UPLOAD_FLAGS

debug_server =
    $PLATFORMIO_CORE_DIR/packages/tool-openocd/bin/openocd
    ${openocd.flags}
```