platformio/platform-ststm32 #296: can't use USB CDC ( SerialUSB)

andhieSetyabudi, 2019-10-10:
please help..
i create my own board using STM32L476R as the main MCU, and i need USB-serial (CDC) to communicate  my PC.
i already set build-flag like shown below
D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC
-D USBCON
-D USBD_VID=0x0483
-D USB_MANUFACTURER=“Unknown”
-D USB_PRODUCT="“test_board”"
-D HAL_PCD_MODULE_ENABLED

and call SerialUSB.begin() in my code.
but there is no serial-port open or connected on my pc

andhieSetyabudi, 2019-10-13:
anyone know where is USBD_Desc defining on ststm32?

andhieSetyabudi, 2019-10-14:
adding peripheral init to set LSE HSE and others in variant.cpp
and it works..
i try to clone some script from disco_l475's variant.cpp file..

closed 2019-10-14 by andhieSetyabudi

fpistm, 2019-10-14:
Hi @andhieSetyabudi 
I guess you get the Nucleo-L476RG as an example for your own board but the `SystemClock_Config` does not initialize the USB clock as there is no USB port per default.
So, up to you to ensure providing the USB clock source at 48MHz.