platformio/platform-ststm32 #300: Help Stm32 bluepill_f103c8 and maple librarys

urosh-dev, 2019-10-17:
Greetings. 
I hope someone can point me to the right direction. I spent 3days with this problem and no luck.
I have blue pill STM32F103c8 and i have a working arduino code that is working if I compile it and upload it with Arduino ide. 
My code:

    #include <Arduino.h>
    #include <Wire_slave.h>
    #define SLAVE_ADDRESS 0x04

    volatile bool receiveFlag = false;
    volatile bool requestFlag = false;
    String toSend;
    uint8_t req_opcode;   // request event register
    uint8_t rcv_opcode;  // receive event register
    uint8_t speed;
    String opcode_str;

    void receiveEvent(int howMany) {
      if (howMany == 0) return;
      for (int i = 0; i < howMany; i++) {
        temp[i] = Wire.read();
        temp[i + 1] = '\0'; //add null after ea. char
      }
      rcv_opcode = temp[0];
      for (int i = 0; i < howMany; ++i)
         temp[i] = temp[i + 1];
      receiveFlag = true;
    }
    void requestEvent()
    {
      requestFlag = true;
      Wire.write(rcv_opcode);
      }

    void setup() {
      // initialize i2c as slave
      Wire.begin(SLAVE_ADDRESS);
      Wire.onReceive(receiveEvent);
      Wire.onRequest(requestEvent);
      Serial.begin(115200);
      Serial.println("Ready!");
      pinMode(PC13, OUTPUT);
    }

    void loop() {
      digitalWrite(PC13, LOW);
      delay(100);
      digitalWrite(PC13, HIGH);
      delay(100);
      if (receiveFlag == true) {
        Serial.println("receive register "+String(rcv_opcode)+".");
        Serial.println("receive_flag "+String(temp)+".");
        receiveFlag = false;
      };
      if (requestFlag ==  true) {
        opcode_str = String("request ") + String(req_opcode) + String(".");
        Serial.println(opcode_str);  
        requestFlag = false;
      }
    }

If I try to compile in platformio I get error:

    In file included from src\main.cpp:2:0:
    C:\Users\uros\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/Wire_slave.h:1:2: error: #error "Something is trying to include Wire_slave.h when Wire.h is already included, they are mutually exclusive"
     #error "Something is trying to include Wire_slave.h when Wire.h is already included, they are mutually exclusive"
      ^~~~~
    src\main.cpp: In function 'void receiveEvent(int)':
    src\main.cpp:20:15: error: 'Wire' was not declared in this scope
         temp[i] = Wire.read();
                   ^~~~
    src\main.cpp: In function 'void requestEvent()':
    src\main.cpp:48:3: error: 'Wire' was not declared in this scope
       Wire.write(rcv_opcode);
       ^~~~
    src\main.cpp: In function 'void setup()':
    src\main.cpp:53:3: error: 'Wire' was not declared in this scope
       Wire.begin(SLAVE_ADDRESS);
       ^~~~
    *** [.pio\build\bluepill_f103c8\src\main.cpp.o] Error 1

The same code works in Arduino IDE.
I also tried to exclude Wire library with lib_ignore = Wire in platformio.ini becouse I see that error is in \Wire/Wire_slave.h. But now I get error:

    Linking .pio\build\bluepill_f103c8\firmware.elf
    .pio\build\bluepill_f103c8\lib931\libWireSlave.a(i2c_slave.c.o): In function `i2c_bus_reset':
    i2c_slave.c:(.text.i2c_bus_reset+0x0): multiple definition of `i2c_bus_reset'
    .pio\build\bluepill_f103c8\FrameworkArduino\libmaple\i2c.c.o:i2c.c:(.text.i2c_bus_reset+0x0): first defined here
    .pio\build\bluepill_f103c8\lib931\libWireSlave.a(i2c_slave.c.o): In function `i2c_init':
    i2c_slave.c:(.text.i2c_init+0x0): multiple definition of `i2c_init'
    .pio\build\bluepill_f103c8\FrameworkArduino\libmaple\i2c.c.o:i2c.c:(.text.i2c_init+0x0): first defined here
    .pio\build\bluepill_f103c8\lib931\libWireSlave.a(i2c_slave.c.o): In function `i2c_master_enable':
    i2c_slave.c:(.text.i2c_master_enable+0x0): multiple definition of `i2c_master_enable'
    .pio\build\bluepill_f103c8\FrameworkArduino\libmaple\i2c.c.o:i2c.c:(.text.i2c_master_enable+0x0): first defined here
    .pio\build\bluepill_f103c8\lib931\libWireSlave.a(i2c_slave.c.o): In function `i2c_master_xfer':
    i2c_slave.c:(.text.i2c_master_xfer+0x0): multiple definition of `i2c_master_xfer'
    .pio\build\bluepill_f103c8\FrameworkArduino\libmaple\i2c.c.o:i2c.c:(.text.i2c_master_xfer+0x0): first defined here
    .pio\build\bluepill_f103c8\lib931\libWireSlave.a(i2c_slave.c.o): In function `_i2c_irq_handler':
    i2c_slave.c:(.text._i2c_irq_handler+0x0): multiple definition of `_i2c_irq_handler'
    .pio\build\bluepill_f103c8\FrameworkArduino\libmaple\i2c.c.o:i2c.c:(.text._i2c_irq_handler+0x0): first defined here
    .pio\build\bluepill_f103c8\lib931\libWireSlave.a(i2c_slave.c.o): In function `_i2c_irq_error_handler':
    i2c_slave.c:(.text._i2c_irq_error_handler+0x0): multiple definition of `_i2c_irq_error_handler'
    .pio\build\bluepill_f103c8\FrameworkArduino\libmaple\i2c.c.o:i2c.c:(.text._i2c_irq_error_handler+0x0): first defined here
    collect2.exe: error: ld returned 1 exit status
    *** [.pio\build\bluepill_f103c8\firmware.elf] Error 1

Here is my platformio.ini file:
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
upload_protocol = stlink
monitor_speed = 115200
monitor_port = COM1
board_build.mcu = stm32f103c8t6
board_build.core = maple
lib_ignore = Wire

Maybe you will ask why i use maple core. Because the wire_slave library works as i expect (using smbus library on raspberrypi), Library Wire doesn't work as expected.

I hope someone can help me. 
Thanks

urosh-dev, 2019-10-17:
I tried with copying librarys from here: 
http://dan.drown.org/stm32duino/STM32F1-2019.10.9.zip
and it works now with some warnings when compileing.

valeros, 2019-10-17:
Hi @urosh-dev! I guess that `multiple definition` errors are fixed in the latest framework version but not yet released. You could try to replace folder `c:\Users\YOUR_USER\.platformio\packages\framework-arduinoststm32-maple\STM32F1` with contents from your archive.

urosh-dev, 2019-10-17:
I did. Thank you!

closed 2019-10-17 by valeros

gpintarelli, 2020-09-23:
Hi!
I'm also trying to use the maple core (rogerclarkmelbourne/Arduino_STM32) with platformIO. I'm having the same problem repported by urosh-dev.

PS1. I also tested the maple core (and the rest of my code) by using it with IDE, as it work like a charm! However, in platmformIO the code won't compile due the multiple definition.
PS2. My platformio.ini is as urosh-dev.

To my knowledge it seems a incompatibility with arduino framework and maple core... However, I don't know to what extent it would be possible to isolate some libraries automatically. 

I am using:
ststm32 -> Version: 8.0.0
Package framework-arduinoststm32-maple -> Version: 2.10000.200103

In addition to the manipulation of folders, would it be possible to work around the problem in patformio.ini itself? Any suggestion is super welcome! Thank you