platformio/platform-ststm32 #301: Could not find a version that satisfies the requirement '>=1.60301.0,<1.80000.0' for your system 'linux_aarch64'

dennislbowman, 2019-10-20:
PIO Core Call Error: "The current working directory /home/dennis/Documents/PlatformIO/Projects/teststm32 will be used for the project.\n\nThe next files/directories have been created in /home/dennis/Documents/PlatformIO/Projects/teststm32\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Could not find a version that satisfies the requirement '>=1.60301.0,<1.80000.0' for your system 'linux_aarch64'"

senceryazici, 2019-12-18:
I'm having the same problem, so far
toolchain-gccarmnoneeabi and tool-dfuutil failed to install when I run 
`pio run --target upload`
But probably all of the remaining packages will fail since majority of them does not support linux_aarch. |'m trying to flash stm32 nucleo board through Jetson Xavier running on ARM CPU

maxgerhardt, 2020-05-21:
Stilll occuring in latest version: See https://community.platformio.org/t/installation-of-toolchain-gccarmnoneeabi-fails-on-arm64-debian-during-pio-platform-install-ststm32/

jnr2820, 2020-06-03:
It would be superb to have the complete toolchain for "aarch64" 

Context:
I ran into this issue while trying to program a Generic STM32F1x Bluepill from a "NanoPi Neo2 Black" running Armbian. These boards have Allwinner H5's and would be great for programing boards with PlatformIO, due to their small size. (They could even be integrated into a project.)
(INFO: http://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO2_Black)

I am using a NanoPi Neo (1) and Duo 2 for now, as they are is "Armv7l" and the toolchain appears to be complete for that architecture.

NanoPi NEO (1) w/Armbian:
`Linux nano-uno-1 5.4.43-sunxi #20.05.1 SMP Sat May 30 20:32:43 CEST 2020 armv7l GNU/Linux`

NanoPi NEO 2 Black w/Armbian:
`Linux nano-cube-black 5.4.43-sunxi64 #20.05.1 SMP Sat May 30 21:02:33 CEST 2020 aarch64 GNU/Linux`

jnr2820, 2020-06-25:
We are still seeing the same issues.

```PackageManager: Installing toolchain-gccarmnoneeabi @ >=1.60301.0,<1.80000.0
Error: Could not find a version that satisfies the requirement '>=1.60301.0,<1.80000.0' for your system 'linux_aarch64'```

And this one: 

```PackageManager: Installing tool-dfuutil @ ~1.9.190708
Error: Could not find a version that satisfies the requirement '~1.9.190708' for your system 'linux_aarch64'```

I'm going to be the squeaky wheel now.......(this is long overdue as this was filed late 2019) 

Hello Mr @ivankravets and Mr @maxgerhardt ,

I have a set of questions: 
1. Is tool chain consistency something that will be addressed with the upcoming repo redesign? If not, is this issue going to be assigned to someone soon?
2. Why is ARM support advertised on the main site when tool chains are not complete for something as common as ststm32 and Atmelsam on "aarch64" hosts? (If anything, it should be marked as "Beta" or "Partial functionality - Actively being developed")
3. Is there documentation somewhere that I managed to miss that says certain platforms are only supported/tested/working on certain host architectures? ***If not, that would probably go a long way to easy frustration.***

Now, while I may be a bit frustrated, I'm willing to also put that energy towards whatever it is I can do to help. Do we need to put together a quick QA script that instructs PIO to go compile a bunch of stuff for different platforms on my ARM64 box so that we can see what is still needs to polishing? Would that be helpful?

valeros, 2020-08-13:
At the moment only GCC v9.3.1 is available for ARM platforms, please try the following configuration:

```
[env:override_default_toolchain]
platform = ststm32
board = ...
framework = ...
platform_packages =
  toolchain-gccarmnoneeabi@~1.90301.0
```

hybotics, 2023-02-10:
I am unable to build for an STM32F407 based board. Specifically, I am trying to build the Marlin Firmware for my 3D Printer with a BigTree Tech GTR 1.0 mainboard.

Error: Could not find the package with 'platformio/toolchain-gccarmnoneeabi @ >=1.60301.0,<1.80000.0' requirements for your system 'linux_aarch64'

Will this ever be fixed?

commit 2429457b (2023-06-14): Switch to GCC v10.3 for Arduino core

Resolves #301, resolves #486, resolves #605, resolves #650
closed 2023-06-14 by valeros