platformio/platform-ststm32 #305: STM 5.7.0.... undefined reference to `ep_def'?

LinoBarreca, 2019-10-24:
New 5.7.0
[...]
Linking .pio\build\XXX\firmware.elf

.pio/build/XXX/FrameworkArduino/stm32/usb/usbd_conf.c.o: In function `USBD_LL_Init':
C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb/usbd_conf.c:520: undefined reference to `ep_def'

Is this a bug or am I doing something wrong?

valeros, 2019-10-24:
Hi @LinoBarreca ! Could you please provide a minimal project to reproduce the issue?

LinoBarreca, 2019-10-24:
@valeros  I am sorry. I don't have a minimal one https://github.com/LinoBarreca/Marlin (branch bugfix-2.0.x)

valeros, 2019-10-24:
What environment do you use?

LinoBarreca, 2019-10-24:
PlatformIO Home 2.3.3 Core 4.1.0b5 on Visual Studio Code 1.39.2
but it's the same as
https://travis-ci.org/MarlinFirmware/Marlin/jobs/602261957

valeros, 2019-10-24:
@LinoBarreca Strange, but your project fails with other error:
```
Marlin/src/module/stepper/trinamic.cpp: In function 'void tmc_serial_begin()':
Marlin/src/module/stepper/trinamic.cpp:296:18: error: 'class TMCMarlin<TMC2209Stepper, 'X', '0', (AxisEnum)0>' has no member named 'beginSerial'; did you mean 'HWSerial'?
         stepperX.beginSerial(115200);
                  ^~~~~~~~~~~
                  HWSerial
Marlin/src/module/stepper/trinamic.cpp:310:18: error: 'class TMCMarlin<TMC2209Stepper, 'Y', '0', (AxisEnum)1>' has no member named 'beginSerial'; did you mean 'HWSerial'?
         stepperY.beginSerial(115200);
                  ^~~~~~~~~~~
                  HWSerial
Marlin/src/module/stepper/trinamic.cpp:324:18: error: 'class TMCMarlin<TMC2209Stepper, 'Z', '0', (AxisEnum)2>' has no member named 'beginSerial'; did you mean 'HWSerial'?
         stepperZ.beginSerial(115200);
                  ^~~~~~~~~~~
                  HWSerial
Marlin/src/module/stepper/trinamic.cpp:345:19: error: 'class TMCMarlin<TMC2209Stepper, 'E', '0', (AxisEnum)3>' has no member named 'beginSerial'; did you mean 'HWSerial'?
         stepperE0.beginSerial(115200);
                   ^~~~~~~~~~~
                   HWSerial
Marlin/src/module/stepper/trinamic.cpp:352:19: error: 'class TMCMarlin<TMC2209Stepper, 'E', '1', (AxisEnum)3>' has no member named 'beginSerial'; did you mean 'HWSerial'?
         stepperE1.beginSerial(115200);
                   ^~~~~~~~~~~
                   HWSerial
compilation terminated due to -fmax-errors=5.
*** [.pio/build/BIGTREE_SKR_PRO/src/src/module/stepper/trinamic.cpp.o] Error 1
```
Any hints?

LinoBarreca, 2019-10-24:
@valeros modify TMCStepper.h
Edit line
`#if (__cplusplus == 201703L) && defined(__has_include)`
into
`#if defined(__has_include)`

at the top of the file. It should compile.

valeros, 2019-10-24:
OK, that helped, but there is another one:
```
In file included from .pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\HAL_softserial_STM32.cpp:27:0:
.pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\HAL_softserial_STM32.h:37:36: error: variable or field 'SoftSerial_Handler' declared void
 extern "C" void SoftSerial_Handler(stimer_t *htim);
                                    ^~~~~~~~
.pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\HAL_softserial_STM32.h:37:36: error: 'stimer_t' was not declared in this scope
.pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\HAL_softserial_STM32.h:37:36: note: suggested alternative: 'timer_t'
 extern "C" void SoftSerial_Handler(stimer_t *htim);
                                    ^~~~~~~~
                                    timer_t
.pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\HAL_softserial_STM32.h:37:46: error: 'htim' was not declared in this scope
 extern "C" void SoftSerial_Handler(stimer_t *htim);
                                              ^~~~
.pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\HAL_softserial_STM32.h:37:46: note: suggested alternative: 'tm'
 extern "C" void SoftSerial_Handler(stimer_t *htim);
                                              ^~~~
                                              tm
.pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\HAL_softserial_STM32.cpp:69:1: error: 'stimer_t' does not name a type; did you mean 'timer_t'?
 stimer_t SSTimerHandle;
 ^~~~~~~~
 timer_t
.pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\HAL_softserial_STM32.cpp: In function 'void HAL_softSerial_init()':
.pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\HAL_softserial_STM32.cpp:74:3: error: 'SSTimerHandle' was not declared in this scope
   SSTimerHandle.timer = SS_TIMER_DEV;
   ^~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\BIGTREE_SKR_PRO\libd55\SoftwareSerialM\HAL_softserial_STM32.cpp.o] Error 1
In file included from .pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\HAL_softserial.h:28:0,
                 from .pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\SoftwareSerial.cpp:40:
.pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\HAL_softserial_STM32.h:37:36: error: variable or field 'SoftSerial_Handler' declared void
 extern "C" void SoftSerial_Handler(stimer_t *htim);
                                    ^~~~~~~~
.pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\HAL_softserial_STM32.h:37:36: error: 'stimer_t' was not declared in this scope
.pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\HAL_softserial_STM32.h:37:36: note: suggested alternative: 'timer_t'
 extern "C" void SoftSerial_Handler(stimer_t *htim);
                                    ^~~~~~~~
                                    timer_t
.pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\HAL_softserial_STM32.h:37:46: error: 'htim' was not declared in this scope
 extern "C" void SoftSerial_Handler(stimer_t *htim);
                                              ^~~~
.pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\HAL_softserial_STM32.h:37:46: note: suggested alternative: 'tm'
 extern "C" void SoftSerial_Handler(stimer_t *htim);
                                              ^~~~
                                              tm
.pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\HAL_softserial_STM32.h:35:71: error: variable or field 'SoftSerial_Handler' declared void
 #define HAL_SOFTSERIAL_TIMER_ISR() extern "C" void SoftSerial_Handler(stimer_t *htim)
                                                                       ^
.pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\SoftwareSerial.cpp:228:1: note: in expansion of macro 'HAL_SOFTSERIAL_TIMER_ISR'
 HAL_SOFTSERIAL_TIMER_ISR() {
 ^~~~~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\HAL_softserial_STM32.h:35:71: error: 'stimer_t' was not declared in this scope
 #define HAL_SOFTSERIAL_TIMER_ISR() extern "C" void SoftSerial_Handler(stimer_t *htim)
                                                                       ^
.pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\SoftwareSerial.cpp:228:1: note: in expansion of macro 'HAL_SOFTSERIAL_TIMER_ISR'
 HAL_SOFTSERIAL_TIMER_ISR() {
 ^~~~~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\HAL_softserial_STM32.h:35:71: note: suggested alternative: 'timer_t'
 #define HAL_SOFTSERIAL_TIMER_ISR() extern "C" void SoftSerial_Handler(stimer_t *htim)
                                                                       ^
.pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerialM\SoftwareSerial.cpp:228:1: note: in expansion of macro 'HAL_SOFTSERIAL_TIMER_ISR'
 HAL_SOFTSERIAL_TIMER_ISR() {
 ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\BIGTREE_SKR_PRO\libd55\SoftwareSerialM\SoftwareSerial.cpp.o] Error 1
```

LinoBarreca, 2019-10-24:
edit platformio.ini

go to [env:BIGTREE_SKR_PRO]
remove SoftwareSerialM=..... from lib_deps and add
lib_ignore = SoftwareSerialM
just below it
(since 5.7.0 includes a SoftwareSerial)

LinoBarreca, 2019-10-24:
(I just pushed the correct platformio.ini to that branch)

valeros, 2019-10-24:
Thank, now it compiles. As for the first issue, I suppose you also need to add `-D HAL_PCD_MODULE_ENABLED` to your build_flags.

LinoBarreca, 2019-10-24:
> 
> 
> Thank, now it compiles. As for the first issue, I suppose you also need to add `-D HAL_PCD_MODULE_ENABLED` to your build_flags.

@valeros  thank you. now it links successfully....but I get a million warnings about redefinition of HAL_PCD_MODULE_ENABLED is this normal or....?

LinoBarreca, 2019-10-24:
@valeros also it now breaks somewhere else when I compile for another environment (STM32F407VE_black) :

`In file included from C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/pinmap.h:22:0,
                 from C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PeripheralPins.h:34,
                 from C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\variants\MARLIN_F407VE/variant.h:35,
                 from C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\system\STM32F4xx/stm32f4xx_hal_conf.h:4,
                 from C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_hal.h:30,
C                 from C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F4xx\Include/stm32f4xx.h:250,
                 from C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\stm32_def.h:34,
ompiling .pio\build\STM32F407VE_black\FrameworkArduino\stm32\dwt.c.o
                 from C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\PortNames.h:33,
                 from C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\PortNames.c:30:
C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:16:12: error: 'PortA' undeclared here (not in a function)
   PA_0  = (PortA << 4) + 0x00,
            ^~~~~
C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:33:12: error: 'PortB' undeclared here (not in a function); did you mean 'PortA'?
   PB_0  = (PortB << 4) + 0x00,
            ^~~~~
            PortA
C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:50:12: error: 'PortC' undeclared here (not in a function); did you mean 'PortB'?
   PC_0  = (PortC << 4) + 0x00,
            ^~~~~
            PortB
C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:68:12: error: 'PortD' undeclared here (not in a function); did you mean 'PortC'?
   PD_0  = (PortD << 4) + 0x00,
            ^~~~~
            PortC
C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:86:12: error: 'PortE' undeclared here (not in a function); did you mean 'PortD'?
   PE_0  = (PortE << 4) + 0x00,
            ^~~~~
            PortD
compilation terminated due to -fmax-errors=5.
*** [.pio\build\STM32F407VE_black\FrameworkArduino\stm32\PortNames.c.o] Error 1
C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.cpp: In function 'uint32_t get_adc_internal_channel(PinName)':
C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.cpp:242:10: error: 'PADC_TEMP' was not declared in this scope
     case PADC_TEMP:
          ^~~~~~~~~
C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.cpp:242:10: note: suggested alternative: 'PADC_BASE'
     case PADC_TEMP:
          ^~~~~~~~~
          PADC_BASE
C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.cpp:257:10: error: 'PADC_VREF' was not declared in this scope
     case PADC_VREF:
          ^~~~~~~~~
C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.cpp:257:10: note: suggested alternative: 'PADC_BASE'
     case PADC_VREF:
          ^~~~~~~~~
          PADC_BASE
C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.cpp:262:10: error: 'PADC_VBAT' was not declared in this scope
     case PADC_VBAT:
          ^~~~~~~~~
C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.cpp:262:10: note: suggested alternative: 'PADC_BASE'
     case PADC_VBAT:
          ^~~~~~~~~
          PADC_BASE
*** [.pio\build\STM32F407VE_black\FrameworkArduino\stm32\analog.cpp.o] Error 1
================================================================================================= [FAILED] Took 8.39 seconds =================================================================================================

Environment              Status    Duration
-----------------------  --------  ------------
megaatmega2560           IGNORED
megaatmega1280           IGNORED
at90usb1286_cdc          IGNORED
at90usb1286_dfu          IGNORED
DUE                      IGNORED
DUE_USB                  IGNORED
DUE_debug                IGNORED
LPC1768                  IGNORED
LPC1769                  IGNORED
melzi                    IGNORED
melzi_optiboot           IGNORED
rambo                    IGNORED
sanguino_atmega644p      IGNORED
sanguino_atmega1284p     IGNORED
STM32F103RE              IGNORED
STM32F103RC_fysetc       IGNORED
STM32F103RC_bigtree      IGNORED
STM32F103RC_bigtree_USB  IGNORED
STM32F4                  IGNORED
STM32F7                  IGNORED
ARMED                    IGNORED
STM32F103VE_longer       IGNORED
mks_robin                IGNORED
mks_robin_lite           IGNORED
mks_robin_mini           IGNORED
mks_robin_nano           IGNORED
jgaurora_a5s_a1          IGNORED
STM32F407VE_black        FAILED    00:00:08.395
BIGTREE_SKR_PRO          IGNORED
BIGTREE_BTT002           IGNORED
teensy31                 IGNORED
teensy35                 IGNORED
STM32F103CB_malyan       IGNORED
esp32                    IGNORED
fysetc_f6_13             IGNORED
linux_native             IGNORED
SAMD51_grandcentral_m4   IGNORED
include_tree             IGNORED`

valeros, 2019-10-24:
@LinoBarreca Looks like this issue is only related to `BIGTREE_SKR_PRO` board as there is indeed an extra `HAL_PCD_MODULE_ENABLED` macro define in `variant.h` that was introduced in [this commit](https://github.com/MarlinFirmware/Marlin/commit/c7acd5c45b2ecd1ba1ceb90e25e3be197f065df3#diff-23ae8b46468be3ce4bae3ab7ca4a1699R268). Are you able to compile the project in Arduino IDE without this issue?

LinoBarreca, 2019-10-24:
@valeros I will be out of country until tuesday. Finally some vacation for me too. I am finishing some other changes to the "NewTimer" branch and I will start looking at it again on oct 29. put this on hold, please. I'll be back next week.

LinoBarreca, 2019-10-28:
@valeros I'm back. I tried to compile with Arduino and the build works. I saw that the `HAL_PCD_MODULE_ENABLED` is` defined on "standard" compile command line.

Here's just one (for simplicity)
```
"c:/users/lino barreca/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../libexec/gcc/arm-none-eabi/8.2.1/cc1plus.exe"
 -quiet
 -I "C:\Users\Lino Barreca\Documents\PlatformIO\Projects\Marlin-bugfix-2.0.x\Marlin"
 -I "C:\Users\Lino Barreca\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\cores\arduino/avr"
 -I "C:\Users\Lino Barreca\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\cores\arduino/stm32"
 -I "C:\Users\Lino Barreca\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\cores\arduino/stm32/LL"
 -I "C:\Users\Lino Barreca\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\cores\arduino/stm32/usb"
 -I "C:\Users\Lino Barreca\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\cores\arduino/stm32/usb/hid"
 -I "C:\Users\Lino Barreca\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\cores\arduino/stm32/usb/cdc"
 -I "C:\Users\Lino Barreca\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/Drivers/STM32F4xx_HAL_Driver/Inc/"
 -I "C:\Users\Lino Barreca\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/Drivers/STM32F4xx_HAL_Driver/Src/"
 -I "C:\Users\Lino Barreca\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/STM32F4xx/"
 -I "C:\Users\Lino Barreca\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc"
 -I "C:\Users\Lino Barreca\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src"
 -I "C:\Users\Lino Barreca\AppData\Local\Arduino15\packages\STM32\tools\CMSIS\5.5.1/CMSIS/Core/Include/"
 -I "C:\Users\Lino Barreca\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/"
 -I "C:\Users\Lino Barreca\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/gcc/"
 -I "C:\Users\Lino Barreca\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\cores\arduino"
 -I "C:\Users\Lino Barreca\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\variants\BLACK_F407XX"
 -I "C:\Users\Lino Barreca\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\libraries\SPI\src"
 -I "C:\Users\Lino Barreca\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\libraries\Servo\src"
 -I "C:\Users\Lino Barreca\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\libraries\IWatchdog\src"
 -I "C:\Users\Lino Barreca\Documents\Arduino\libraries\U8glib\src"
 -I "C:\Users\Lino Barreca\Documents\Arduino\libraries\TMCStepper\src"
 -I "C:\Users\Lino Barreca\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\libraries\SoftwareSerial\src"
 -imultilib thumb/v7e-m+fp/hard
 -iprefix "c:\users\lino barreca\appdata\local\arduino15\packages\stm32\tools\arm-none-eabi-gcc\8.2.1-1.7\bin\../lib/gcc/arm-none-eabi/8.2.1/"
 -isysroot "c:\users\lino barreca\appdata\local\arduino15\packages\stm32\tools\arm-none-eabi-gcc\8.2.1-1.7\bin\../arm-none-eabi"
 -MMD C:\Users\LINOBA~1\AppData\Local\Temp\arduino_build_977871\sketch\src\sd\SdFatUtil.cpp.d
 -MQ C:\Users\LINOBA~1\AppData\Local\Temp\arduino_build_977871\sketch\src\sd\SdFatUtil.cpp.o 
 -D__USES_INITFINI__ -D STM32F4xx -D ARDUINO=10805 -D ARDUINO_BLACK_F407ZG -D ARDUINO_ARCH_STM32 -D "BOARD_NAME=\"BLACK_F407ZG\""
 -D STM32F407xx -D USBCON -D USBD_VID=0x0483 -D "USB_MANUFACTURER=\"Unknown\"" -D "USB_PRODUCT=\"BLACK_F407ZG\""
 -D HAL_PCD_MODULE_ENABLED
 -D USBD_USE_CDC
 -D DISABLE_GENERIC_SERIALUSB 
 -D HAL_UART_MODULE_ENABLED
C:\Users\LINOBA~1\AppData\Local\Temp\arduino_build_977871\sketch\src\sd\SdFatUtil.cpp
 -quiet -dumpbase SdFatUtil.cpp -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -march=armv7e-m+fp 
 -auxbase-strip C:\Users\LINOBA~1\AppData\Local\Temp\arduino_build_977871\sketch\src\sd\SdFatUtil.cpp.o -O1 -Wall -Wextra
 -std=gnu++14 -ffunction-sections -fdata-sections -fno-threadsafe-statics -fno-rtti -fno-exceptions -fno-use-cxa-atexit --param max-inline-insns-single=500
-o C:\Users\LINOBA~1\AppData\Local\Temp\ccdPCF37.s
```

valeros, 2019-10-28:
@LinoBarreca Could you please upload the entire build log from Arduino IDE to https://pastebin.com ? I see a bunch of strange flags like `-march=armv7e-m+fp` where do they come from?

LinoBarreca, 2019-10-29:
@valeros Arduino 1.8.5 puts them on command line.
I didn't take the one above from a build log but directly from the process command line during build.
A full clean/build for just SKR_Pro on Arduino takes **a life** (hours, compared to about a minute and 10 seconds on vscode)

But I saw that even the other "variant" has the same problem (with a different files' logic)
Look [here](url=https://github.com/MarlinFirmware/Marlin/pull/15655/commits/a7445c129b7f3e11c8f8b1d100eb85c4328736eb#diff-0aa249bd3c78b9b8f76f74af3e43df43)

This is a slightly different situation from SKR_Pro. The HAL_PCD_MODULE_ENABLED was configured with a define in hal_conf_custom.h but at the time of linking I had the same problem of the same undefined ep_def.

If I put it in the platformio.ini everything works. To avoid redefinition warnings I had to take it away from the variants, I prefer to not disable it globally.

by the way...today I did an update of the framework-arduinostm32 v3.10700.191001 to v3.10700.191028 and wanted to check if they fixed it somehow.
Now something weird happens:

- I removed HAL_PCD_MODULE_ENABLED from platformio.ini
- I enabled it back again in the variant.h
- Clean & Build
Now I get a redefinition warning as if I never removed it from platformio.ini.
`*** [.pio\build\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_uart_ex.c.o] Build interrupted.
*** [.pio\build\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_spi.c.o] Build interrupted.
*** [.pio\build\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_usart_ex.c.o] Build interrupted.
*** [.pio\build\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_tim.c.o] Build interrupted.
*** [.pio\build\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_uart.c.o] Build interrupted.
*** [.pio\build\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_tim_ex.c.o] Build interrupted.
*** [.pio\build\BIGTREE_SKR_PRO\FrameworkArduino\stm32\HAL\stm32yyxx_hal_usart.c.o] Build interrupted.
In file included from C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\system\STM32F4xx/stm32f4xx_hal_conf.h:4:0,
                 from C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_hal.h:30,
                 from C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Src/stm32f4xx_hal_wwdg.c:117,
                 from C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\HAL\stm32yyxx_hal_wwdg.c:14:
C:\Users\Lino Barreca\.platformio\packages\framework-arduinoststm32\variants\BIGTREE_GENERIC_STM32F407_5X/variant.h:264:0: warning: "HAL_PCD_MODULE_ENABLED" redefined
 #define HAL_PCD_MODULE_ENABLED //NOTE: THIS IS REQUIRED but if enabled here (and not in platformio.ini) platformio core <4.1.0 makes the linking fail
 
<command-line>:0:0: note: this is the location of the previous definition`

valeros, 2019-10-29:
The latest `v3.10700.191028` now automatically appends `HAL_PCD_MODULE_ENABLED` if any type of USB is enabled (as in Arduino IDE).

closed 2019-10-29 by LinoBarreca