platformio/platform-ststm32 #308: "Cutom Board" never shows up

wz233, 2019-10-28:
To whom it may concern,

I follow the instructions listed in [Custom Embedded Board](https://docs.platformio.org/en/latest/platforms/creating_board.html), and use command "`platformio boards`" to check whether the custom board I created is listed, but the board I created never shows up. The board description JSON file is listed below:

```json
{
  "build": {
    "cpu": "cortex-m4",
    "extra_flags": "-DSTM32F412Rx",
    "f_cpu": "100000000L",
    "mcu": "stm32f412ret6"
  },
  "frameworks": [
    "mbed",
    "stm32cube"
  ],
  "platform": [
    "ststm32"
  ],
  "debug": {
    "default_tools": [
      "stlink"
    ],
    "jlink_device": "STM32F412RE",
    "onboard_tools": [
      "stlink"
    ],
    "svd_path": "STM32F41x.svd"
  },
  "name": "General-F412RE",
  "upload": {
    "maximum_ram_size": 262144,
    "maximum_size": 524288,
    "protocol": "stlink",
    "protocols": [
      "jlink",
      "stlink",
      "blackmagic",
      "mbed"
    ]
  },
  "url": "https://developer.mbed.org/platforms/ST-Nucleo-F411RE/",
  "vendor": "SVV"
}

```

ivankravets, 2019-10-28:
```
  "platform": [
    "ststm32"
  ],
```

If your board is compatible with a few platforms, use `platforms: []`, S at the end, if with one, just `platform: "ststm32`.

wz233, 2019-10-28:
Dear Ivan,

I modified it, and reopen PlatformIO, it shows following error:

[image]

Unless I changed to `"platform": ["ststm32"]`, but the board still doesn't show up.

wz233, 2019-10-28:
I tried and found the problem, the microcontroller i use is "STM32F412RET6" which is not in PlatformIO's device list, thus it will never display.

ivankravets, 2019-10-28:
Ok, please attach here a new JSON. Where did you place it?

ivankravets, 2019-10-28:
@valeros please help

wz233, 2019-10-28:
Check the JSON below:

```json
{
  "build": {
    "cpu": "cortex-m4",
    "extra_flags": "-DSTM32F412Rx",
    "f_cpu": "100000000L",
    "mcu": "stm32f412ret6"
  },
  "platform": "ststm32",
  "debug": {
    "default_tools": [
      "stlink"
    ],
    "jlink_device": "STM32F412RE",
    "onboard_tools": [
      "stlink"
    ],
    "openocd_board": "st_nucleo_f4",
    "svd_path": "STM32F41x.svd"
  },
  "frameworks": [
    "mbed",
    "stm32cube"
  ],
  "name": "ST Nucleo F412RE",
  "upload": {
    "maximum_ram_size": 262144,
    "maximum_size": 524288,
    "protocol": "stlink",
    "protocols": [
      "jlink",
      "stlink",
      "blackmagic",
      "mbed"
    ]
  },
  "url": "https://developer.mbed.org/platforms/ST-Nucleo-F411RE/",
  "vendor": "ST"
}
```

> I just modified it from `nucleo_f412zg.json`

wz233, 2019-10-28:
I've no idea what happened, I changed back to "stm32f412ret6", close the PIO home and reopened it, now it works! Thanks for Ivan's support!

closed 2019-10-28 by wz233

ivankravets, 2019-10-28:
Happy coding with PlatformIO!

wz233, 2019-10-28:
Hi, Ivan,

I tried to add new board on a new computer with a clean Win 10 and verify the weird problem I described above again, and finally got a solution about "how to add a new board".

- Copy a board description JSON file with similar MCU of your own board located in "`%HOME%\.platformio\platforms\ststm32\boards`", I chose "`nucleo_f412zg.json`".
- Rename the copied file to the name you want, take "MyBoard_F412RE" for example;
- Open VSCode, run `pio boards` command to refresh the board list, you should be able to see the board with name "MyBoard_F412RE";
- Go back to "`%HOME%\.platformio\platforms\ststm32\boards`" and modify necessary parts of `MyBoard_F412RE.json` file based on your hardware configuration.

> NOTE: If you modify the board description JSON file directly without running `pio boards` ahead, you will get the error message described below.
> Dear Ivan,
> 
> I modified it, and reopen PlatformIO, it shows following error:
> 
> [image]
> 
> Unless I changed to `"platform": ["ststm32"]`, but the board still doesn't show up.