platformio/platform-ststm32 #310: mbed build API internal error

hpoduszczak, 2019-11-06:
PIO Core Call Error: "The current working directory /Users/hop/Documents/PlatformIO/Projects/FulminoDiscovery will be used for the project.\n\nThe next files/directories have been created in /Users/hop/Documents/PlatformIO/Projects/FulminoDiscovery\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\nPlatformManager: Installing ststm32\nststm32 @ 5.7.0 has been successfully installed!\nPackageManager: Installing toolchain-gccarmnoneeabi @ >=1.60301.0,<1.80000.0\ntoolchain-gccarmnoneeabi @ 1.70201.0 is already installed\nThe platform 'ststm32' has been successfully installed!\nThe rest of packages will be installed automatically depending on your build environment.\n\n\nError: Processing disco_f072rb (platform: ststm32; board: disco_f072rb; framework: mbed)\n--------------------------------------------------------------------------------\nPackageManager: Installing framework-mbed @ ~5.51401.0\nframework-mbed @ 5.51401.191023 has been successfully installed!\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/ststm32/disco_f072rb.html\nPLATFORM: ST STM32 5.7.0 > ST Discovery F072RB\nHARDWARE: STM32F072RBT6 48MHz, 16KB RAM, 128KB Flash\nDEBUG: Current (stlink) On-board (stlink) External (blackmagic, jlink)\nPACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-mbed 5.51401.191023 (5.14.1)\nCollecting mbed sources...\nwrite() takes 2 positional arguments but 3 were given\nmbed build API internal error\n========================= [FAILED] Took 286.47 seconds ========================="

ivankravets, 2019-11-06:
Do you use the latest dev/platform? `pio update`.

samiralavi, 2020-06-21:
I have faced the same error after defining a custom target for mbed. It seems to be replicable by creating a custom target for mbed. 

Here is the build output:

Executing task: platformio run --verbose <

Processing amir_board (platform: ststm32; board: amir_board; framework: mbed; build_flags: -DLV_CONF_INCLUDE_SIMPLE)

CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/amir_board.html
PLATFORM: ST STM32 6.1.1 > ST 32F746G AMIR
HARDWARE: STM32F746NGH6 216MHz, 320KB RAM, 1MB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, jlink)
PACKAGES: 
 - framework-mbed 6.51401.200402 (5.14.1) 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Collecting mbed sources...
Detected custom target file
mbed build API internal error
write() takes 2 positional arguments but 3 were given
=============================================================================== [FAILED] Took 0.73 seconds ===============================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

samiralavi, 2020-06-21:
I found the bug here:
line 82 of builder-framework-mbed/pio_mbed_adapter.py

`sys.stderr.write("Failed to extract info for %s target\n", self.target)`

The write function is not like printf and only accepts one argument.

ivankravets, 2020-06-22:
Fixed! Please run `pio update` command.

closed 2020-06-22 by ivankravets

araujorayza, 2024-09-02:
hi, @ivankravets. I'm having this issue. the pio update command has not solved it. 

Here's the build output
```
 *  Executing task: platformio run 

Processing llc_board (platform: ststm32; board: llc_board; framework: mbed)
--------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/llc_board.html
PLATFORM: ST STM32 (17.5.0) > LLC Board (custom board based on Nucleo F746ZG)
HARDWARE: STM32F746ZGT6 216MHz, 320KB RAM, 1MB Flash
PACKAGES: 
 - framework-mbed @ 6.61700.231105 (6.17.0) 
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
Collecting mbed sources...
Detected custom target file
target LLC_BOARD has an incomplete target definition
mbed build API internal error
========================================= [FAILED] Took 0.37 seconds =========================================

 *  The terminal process "platformio 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

```