platformio/platform-ststm32 #314: Platform-ststm32 refuses to compile

emilhse, 2019-11-13:
i have updated everything but still getting this error please help

xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-11:28)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
 
embedded:startup.tcl:26: Error: Can't find scripts/interface/stlink.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 26
*** [upload] Error 1

emilhse, 2019-11-13:
i really tried everything i could please help me. i tried  https://github.com/platformio/platform-ststm32/issues/266#issuecomment-526292630 . i tried to install new versions 
like ststm32@5.7.0 no luck . Please please please help

emilhse, 2019-11-13:
[image]
I tried almost everything i could

emilhse, 2019-11-13:
BIGTREE_SKR_MINI          IGNORED
STM32F4                   IGNORED
STM32F7                   IGNORED
ARMED                     IGNORED
alfawise_U20              IGNORED
mks_robin                 IGNORED
mks_robin_lite            IGNORED

mks_robin_mini            IGNORED
mks_robin_nano            IGNORED
jgaurora_a5s_a1           IGNORED
black_stm32f407ve         IGNORED
BIGTREE_SKR_PRO           FAILED    00:01:08.477
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
malyanm200                IGNORED
esp32                     IGNORED
fysetc_f6_13              IGNORED
linux_native              IGNORED
adafruit_grandcentral_m4  IGNORED
==================== 1 failed, 0 succeeded in 00:01:08.477 ====================

ivankravets, 2019-11-13:
Please delete all ststm32 dev/platforms and use the latest.

emilhse, 2019-11-13:
I have deleted everything and made fresh new install of platformio now im getting this error
[image]

fpistm, 2019-11-14:
Hi @emilhse 
You should try to update STLink firmware using the STM32CubeProgrammer.

ivankravets, 2019-11-14:
It seems like you forgot to install ST-Link drivers. See https://docs.platformio.org/en/latest/plus/debug-tools/stlink.html#drivers

emilhse, 2019-11-14:
i have installed the drivers still getting this error

xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-11:28)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
 

hla_swd
Error: open failed
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked

emilhse, 2019-11-14:
I installed STM32CubeProg software too and tested (tried to compile again) after installation still no luck

fpistm, 2019-11-14:
Did you update your stlink firmware ? "Firmware upgrade" button.

emilhse, 2019-11-14:
yes i did

emilhse, 2019-11-14:
i says No ST-Link device detected

emilhse, 2019-11-14:
My board is Bigtreetech SkR pro firmware is upgrated via sd card

ivankravets, 2019-11-14:
@emilhse does STM32CubeProg see your ST-Link probe? Is it original or clone?

emilhse, 2019-11-14:
I don't have stm32cubeprog. All I have is bigtreetech skr pro v1.1 3d printer board. I need to compile bin and then load it via SD card

ivankravets, 2019-11-14:
1. This issue is not related to PlatformIO
2. You don't want to hear @fpistm or me
3. See https://www.youtube.com/watch?v=5RSy7xAoonU and try to install ST-Link drivers and their STM32CubeProgrammer just to UPGRADE FIRMWARE ON ST-LINK, not on your printer.

closed 2019-11-14 by ivankravets

emilhse, 2019-11-14:
i will try
the thing is the platformio even creating firmware.elf. i will try to convert elf to bin

emilhse, 2019-11-14:
i think i found the issue i need to find where its described common flags ENABLE_USB_MASS_STORAGE

emilhse, 2019-11-14:
> 1. This issue is not related to PlatformIO
> 2. You don't want to hear @fpistm or me
> 3. See https://www.youtube.com/watch?v=5RSy7xAoonU and try to install ST-Link drivers and their  just to UPGRADE FIRMWARE ON ST-LINK, not on your printer.

is there any way to put this flag? because platformio creates already elf file all i need is that it generates .bin instead

emilhse, 2019-11-14:
i found st link v 2 and connected it i have read the bin file on chip. when i tried to compile after it dives me this error

hla_swd

Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: open failed
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked
 
*** [upload] Error 1
Unable to execute: platformio
Make sure cmd:'platformio' and cwd:'C:\Users\emilio\Desktop\Marlin-bugfix-2.0.x' exists and have correct access permissions.
Binaries are found in these folders: C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\
Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files\Git\cmd;C:\Program Files\LLVM\bin;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utili
ty\ST-LINK Utility;C:\Users\emilio\AppData\Local\Programs\Python\Python38\Scripts\;C:\Users\emilio\AppData\Local\Programs\Python\Python38\