platformio/platform-ststm32 #315: How to use C++ with the STMCube framework

scottc11, 2019-11-15:
I'm trying to figure this out now, but it would be great if someone with experience could help me out.  

I understand that the `HAL` framework supplied by STM is compatible with C++, but whenever I try to write C++ in my cube projects I get build errors etc.

I tried renaming my `main.c` file to `main.cpp`, which will compile and upload the code to the board, but the MCU will not operate (I think it has something to do with the clock). Anyways... anyone else using C++ in their STMCube platformIO projects?

valeros, 2019-11-20:
Hi @scottc11 ! Could you please provide more information? What build errors do you see? Does your code work if you use ony `C` language?

matte86, 2019-11-27:
Hi guys,

I think I am having the same issue so I write here.

I am developing on an STM32 Nucleo F401RE. I downloaded the hal-blinky example and everything works fine. My `platformio.ini` file looks like this:

[env:nucleo_f401re]
platform = ststm32
framework = stm32cube
board = nucleo_f401re
build_flags = -DF4
upload_protocol = stlink

I noticed that my 'c_cpp_properties.json' contains the following line:

"compilerPath": "\"/Users/matteoagnelli/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-gcc\" -mthumb -mcpu=cortex-m4"

So I assume it is using the gcc compiler.

If I try renaming the main.c to main.cpp and wrap the blinky example code into a class everything compiles fine but the LED does not blink anymore; I think this is the same behavior as reported by @scottc11 .

Is there a way to configure platformio with the  'framework = stm32cube' to use the arm-none-eabi-g++ compiler? Or perhaps I am looking in a different direction?

matte86, 2019-11-28:
Update: it is now working for me. I am not an expert of embedded programming myself so perhaps I was doing something wrong.

I took the example at [https://github.com/iancanada/STM32Cube-Cpp-programming-example], readapted it for my board and everything seems to work properly.

scottc11, 2019-12-03:
> If I try renaming the main.c to main.cpp and wrap the blinky example code into a class everything compiles fine but the LED does not blink anymore

@matte86 ya thats what was happening when I tried this. It seemed to build and upload correctly, but the program would not execute on the chip.

So I am reading the example project you mentioned and much like all the other examples I have found scattered online it uses the Keil IDE. Are you saying you got that project working using PIO + VSCode?

matte86, 2019-12-03:
Hey @scottc11, correct. I got the example to compile and run properly with PIO + VSCode.
I am not an expert embedded programmer myself but I found out that the way I define the SysTick_Handler function impacts the proper execution of the code. For example, if I move that function into the main.cpp file the program does not seem to work (LED is not blinking).

scottc11, 2019-12-07:
@matte86 mind sharing a bit of your code?

manuelbl, 2019-12-11:
I'm successfully using STMcube with C++. Except for renaming your files from `.c` to `.cpp`, there's one major thing that you need to do:

Whenever you override a function of the framework, you have to declare it as `extern "C"`, e.g.:

```
extern "C" void HAL_UART_TxCpltCallback(UART_HandleTypeDef *huart) {
    ...
}
```

This applies to:
- interrupt handlers such as `EXTI0_IRQHandler`, `TIM3_IRQHandler`
- callback functions such as `HAL_SPI_TxCpltCallback`, `HAL_TIM_TriggerCallback`
- `MspInit`/`MspDeinit` functions such as `HAL_SPI_MspInit`, `HAL_I2S_MspInit`

In short, it applies to everyting that has a default implementation using *weak* linkage and can be replaced with a custom implementation.

If you don't declare it as `extern "C"`, the C++ compiler uses a naming and calling convetion incompatible with the HAL library and the linker will wire up your program such that the default implementations instead of your code is called.