platformio/platform-ststm32 #318: Disabling reset in .cfg, successful programming?

silardg, 2019-11-17:
Hi!

I have  been debugging my board (using cmwx1zzabz-093) all day, couldn't get it to program.
After hours of googling and looking and forums I have realized that in the config file (stm32l0discovery.cfg, openocd directory), all I had to do is set `reset_config none` .

Could this be, because I'm using a different ST-LINK, than provided with the B-L072Z-LRWAN1 ? I still have the same connections between my board and the ST-LINK as stated in the B-L072Z schematic.

Wanted to open this issue for people that use the cmwx1zzabz-093 without the on-board ST-LINK.

valeros, 2019-11-20:
Hi @silardg ! Thanks for the info! So what do we need to do? I'm not sure we can cover all possible combinations of hardware and debug probes, but you can configure OpenOCD extra commands as for example in [bluepill_f103c6.json](https://github.com/platformio/platform-ststm32/blob/develop/boards/bluepill_f103c6.json#L27).

silardg, 2019-11-20:
Yes something like that, but I would prefer it to have it avaliable in the platformio.ini file, that way it is easy to configure the board if it's not used with the dev board.

silardg, 2019-11-20:
Also I simple note would be useful too, which just informs you that you can modify the config file if you are having troubles with reseting it.

fpistm, 2019-11-20:
Did you try to update the STLink firmware with the STM32CubeProgrammer?