platformio/platform-ststm32 #320: STM32F103RET6 not work  blink programm

ESAlexhi, 2019-11-24:
Hello! 
I try to use STM32F103RET6 in platformio , the program  write over st-link  ОК, but does not work. What can be wrong?
My programm:

const int LEDPROG = PC13;     
void setup() {
pinMode(LEDPROG,OUTPUT);
}

void loop() {
digitalWrite(LEDPROG,LOW);delay(1000);
digitalWrite(LEDPROG,HIGH);delay(1000);
}

My  platformio.ini:

[env:genericSTM32F103RE]
platform = ststm32
board = genericSTM32F103RE
board_build.mcu = stm32f103ret6
board_build.f_cpu = 72000000L
framework = arduino
upload_protocol =stlink

valeros, 2019-11-26:
There are a few possible reasons why this doesn't work. Is there a bootloader in flash memory?

ESAlexhi, 2019-11-26:
I do not use bootloader, I just flash this program through the St-link.

nopnop2002, 2020-01-15:
Probably the problem with the maple core toolchain.
Add board_build.core=stm32 to platformio.ini
Add #include <Arduino.h> to main.cpp

	[env:genericSTM32F103RE]
	platform = ststm32
	board = genericSTM32F103RE
	board_build.mcu = stm32f103ret6
	board_build.f_cpu = 72000000L
	framework = arduino
	upload_protocol =stlink
	board_build.core=stm32

	#include <Arduino.h>
	const int LEDPROG = PC13;
	void setup() {
	pinMode(LEDPROG,OUTPUT);
	}

	void loop() {
	digitalWrite(LEDPROG,LOW);delay(1000);
	digitalWrite(LEDPROG,HIGH);delay(1000);
	}