platformio/platform-ststm32 #325: Can't upload to board without manual reset using ST-LINK V2

scottc11, 2019-12-07:
Really having a tough time hear 🤔. I was uploading to my `bluepill_f103c8` boards just fine for over a month, then something happened... Can't figure it out. I have tried two different st-link programmers like [this one](https://www.adafruit.com/product/2548) both with the same result.

I CAN upload to the board somewhat successfully if I carefully time pressing the **reset button** on the bluepill board, but this only uploads the program - it then fails with `target halted due to debug-request` which won't allow me to debug the program.

Any ideas what could have happened hear? I re-installed PIO to try and fix it with no luck. Theres a chance PIO automatically upgraded itself prior to this which could have caused the issue but I can't quite remember. 

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8_128k.html
PLATFORM: ST STM32 5.7.0 > BluePill F103C8 (128k)
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 128KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-stm32cube 2.0.181130, tool-stm32duino 1.0.1, tool-openocd 2.1000.190707 (10.0), tool-dfuutil 1.9.190708
Warning! Cannot find a linker script for the required board! Firmware will be linked with a default linker script!
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio/build/bluepill_f103c8_128k/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [          ]   0.2% (used 44 bytes from 20480 bytes)
PROGRAM: [          ]   2.1% (used 2692 bytes from 131072 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, dfu, jlink, stlink
CURRENT: upload_protocol = stlink
Uploading .pio/build/bluepill_f103c8_128k/firmware.elf
xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-15:21)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

hla_swd
none separate

Error: init mode failed (unable to connect to the target)
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked

*** [upload] Error 1
```

DAVIDBETAN2000, 2020-10-30:
same problem
Uploading .pio\build\bluepill_f103c8_128k\firmware.elf
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-09:29)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

hla_swd
none separate

SenvenQi, 2021-05-22:
same problem
Uploading .pio\build\bluepill_f103c8_128k\firmware.elf
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-09:29)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

hla_swd
none separate